shadowsocks/openwrt-shadowsocks #31: 该如何做到每个json配置和SS进程?

mindcat, 2015-01-03:
我看到/etc/shadowsocks/config.json只能放置一个服务器配置. 那么如果我有多个服务器配置该怎么做? LuCI那边也是只能放置一个.

aa65535, 2015-01-03:
客户端只能一次只能连接一个服务端。
多进程因为涉及到负载均衡，所以默认是不提供多进程方案的。

closed 2015-01-03 by aa65535

mindcat, 2015-01-03:
为何这种事情都得要自己亲手做喵! QAQ

aa65535, 2015-01-03:
因为**负载均衡**不是由 shadowsocks 来负责的喵~~

mindcat, 2015-01-03:
@aa65535 不... 猫猫想要的是随意开关指定的SS代理配置喵! 本地端口每个SS代理都不一样喵! 喵....

mindcat, 2015-01-03:
```bash
╭─mindcat@mindcat-linux-pc ~/mydev/pual_bot  ‹master*› 
╰─➤  systemctl | grep shadows                                                                                                                                   130 ↵
grep: warning: GREP_OPTIONS is deprecated; please use an alias or script
shadowsocks@blackhole.service                                                            loaded active running   Shadowsocks Client Service
shadowsocks@mengsky.service                                                              loaded active running   Shadowsocks Client Service
shadowsocks@shadowcheap.service                                                          loaded active running   Shadowsocks Client Service
system-shadowsocks.slice                                                                 loaded active active    system-shadowsocks.slice
```
```
╭─mindcat@mindcat-linux-pc ~/mydev/pual_bot  ‹master*› 
╰─➤  ps -e | grep sslocal
grep: warning: GREP_OPTIONS is deprecated; please use an alias or script
21635 ?        00:00:00 sslocal
25300 ?        00:00:00 sslocal
25313 ?        00:00:00 sslocal
```
这样的赶脚喵?

mindcat, 2015-01-03:
[image]
表示猫猫只好亲手改init文件了喵!

aa65535, 2015-01-03:
对于这种不常见的需要，只能自己动手解决了。