shadowsocks/openwrt-shadowsocks #32: ipv6 support

lsylsy2, 2015-01-09:
现在openwrt的shadowsocks不支持IPv6；
至少已经发现：在ss-rule的处理语句是IPv4的，在config.json内直接输入IPv6地址会被判断为域名；其他的还没有太关注（我自己用shadowvpn，帮别人配置的）
另外，在resolve_ip的函数内，有没有想过ISP封禁了ping的人民感受……

aa65535, 2015-01-10:
resolve_ip 函数功能有限，如果不介意额外安装一个 resolveip 包的话，是可以的。

cokebar, 2015-01-10:
顺道问一下 如果ss-redir通过IPv4连接到服务器 此时想通过ss-redir访问IPv6网站 可能么 如何配置? 之前自己试着加了几条ip6tables 没搞出来。
服务端和客户端两侧都有原生IPv6，不过IPv6访问国外实在缓慢但是如果IPv6存在会导致优先走IPv6 此时google啥的就不走ss了 略慢；但又不想舍弃IPv6 因为想挂PT下载

aa65535, 2015-01-11:
没有 IPv6 环境，无法测试，这种需求可能无法只通过 ip6tables 规则做到。

另外， 因为 CHNRoute 里都是 IPv4 地址，在 IPv6 环境下没必要使用 ss-rules.

bao3, 2015-01-24:
题主，这逻辑有点怪怪的，都已经IPv6了，还需要 ss或者还需要 区分路由吗？干脆就不用这么折腾了吧

lsylsy2, 2015-01-24:
@aa65535 @bao3 需求是，client和server之间通过IPv6连接，但是透过shadowsocks访问的是IPv4网站。
考虑个极端情况，client只有V6地址没有V4，显然就有必要做了；
实际情况是教育网的IPv4“免费地址”为白名单，不在白名单IP内的所有国外IP均无法连接。

AustinChou, 2015-02-01:
ss-rules直接使用IPv6地址会报错，但是使用域名来连接就不会。应该是resolveip的问题。
目前解决这个问题的方式就是新建二级域名，并添加AAAA纪录。

aa65535, 2015-02-01:
@AustinChou 
`ss-rules` 目前不支持 IPv6 的，因为最终 IPv6 地址还是会被过滤掉，可能会导致与服务器连接时的 loopback 。所以在 [resolveip](https://github.com/shadowsocks/openwrt-shadowsocks/blob/v2.1.2/files/shadowsocks.rule#L183) 上添加了 `-4` 这个参数。

AustinChou, 2015-02-01:
@aa65535 
我知道ss-rules 目前只支持 IPv4。我的意思是 ShadowSocks over IPv6，也就是 config.json 中 `server_name` 处填写 IPv6 地址。这样的话 `ss-rules` 会报错。某种情况下这样做是有意义的（教育网 IPv6 不限制流量，不限速，不收取费用）

aa65535, 2015-02-01:
@AustinChou 因为 IPv6 地址会在 [这里](https://github.com/shadowsocks/openwrt-shadowsocks/blob/v2.1.2/files/shadowsocks.rule#L194) 被过滤掉，确定这样的话不会造成 loopback ?
虽然 OpenWrt 默认没有 ip6tables 的 nat 表。

closed 2015-02-01 by aa65535

LazyZhu, 2015-02-01:
Why not using openwrt bb's native command `validate_data` to test the server_name:
validate_data ipaddr server_name

PS: openwrt bb fully support IPv6 NAT (kernel >3.7.0).

aa65535, 2015-02-01:
@LazyZhu 
I want the ss-rules can work in other linux platform.
For now, it works well when using IPv6, so I have no plans to improve the IPv6 NAT.

commit 71260e2f (2015-02-01): fix #32