shadowsocks/openwrt-shadowsocks #35: optimize iptables NAT table?

dawncold, 2015-01-11:
看着条数挺多的，用一段时间后会有一些统计数据出来，比如哪部分的国内IP经常被访问，本着经常被访问的有可能你马上又会访问的原则，可以给这些规则排排序，加个定时任务，每天或者每隔一段时对规则排排序，也许能加快零点几毫秒（没有做过测试）。

`iptables -t nat -xvnL SS_SPEC_WAN_AC | tail -n+12 | head -n -1 | sort -rnsk 1,1 | awk '{ print $NF }' > /etc/shadowsocks/ignore.list`

从第四行开始显示是因为前面两行是
`
Chain SS_SPEC_WAN_AC (2 references)
pkts      bytes target     prot opt in     out     source               destination
`类似这样的信息，第三行是shadowsocks服务器的位置，后面几行是固定保留IP地址段，而最后一行不显示是默认走shadowsocks的那条规则，最终会打印出最后一列，也就是IP段，覆盖掉`/etc/shadowsocks/ignore.list`后重启一下就生效了。

aa65535, 2015-01-12:
没必要这么复杂， 安装 ipset 就行了。
iptables 本身不会有性能问题，问题在于 match 效率。

closed 2015-01-12 by aa65535

dawncold, 2015-01-12:
@aa65535 ipset可以放那么多IP进去么？

aa65535, 2015-01-12:
@dawncold `hash:net` 是可以的。
如果系统中有安装 ipset ，ss-rules 会优先使用 --match-set 方式，避免因为过多的 SS_SPEC_WAN_AC 影响性能。

dawncold, 2015-01-12:
@aa65535 哦，原来这样，那天用ipset加IP最多加到65535，谢谢了。