shadowsocks/openwrt-shadowsocks #36: 希望实现针对端口的透明代理（如仅对1024一下的保留端口执行透明代理）

gcell, 2015-01-11:
如题，日常应用中80和443端口已经足够，考虑到ssh等应用，系统保留端口也足够，对这些端口执行透明代理，其他端口正常出入，这样的策略更符合使用规律。一般BT都是通过端口映射占用后面的端口，简单区分一下可有效防止家用电脑BT流量占用ss通道，对一些非常规应用的兼容性也将有所提升。
如能实现，感激不尽！
非常感谢开发者带来这么好用的软件！

aa65535, 2015-01-12:
对于你这种需求可以自己通过添加 `-e` 参数来解决
修改 /etc/init.d/shadowsocks 的[这个位置](https://github.com/shadowsocks/openwrt-shadowsocks/blob/e0d8985/files/shadowsocks.spec#L77) ，添加 `-e "-m multiport --dports 22,80,443"` ：
```sh
	/usr/bin/ss-rules \
		-c "$CONFIG_FILE" \
		-i "$ignore_list" \
		-a "$ac_args" \
		-b "$wan_bp_ip" \
		-w "$wan_fw_ip" \
		-e "-m multiport --dports 22,80,443" \
		-o
	return $?
```

closed 2015-01-12 by aa65535

gcell, 2015-01-12:
真是太快了，非常感谢！

amoboy, 2015-03-07:
请问现在
	/usr/bin/ss-rules \
		-c "$CONFIG_FILE" \
		-i "$ignore_list" \
		-a "$ac_args" \
		-b "$wan_bp_ip" \
		-w "$wan_fw_ip" \
 		-e "$ipt_ext" \
		-e "-m multiport --dports 22,80,443"\
多一个 -e "$ipt_ext" \
这个要怎么处理

aa65535, 2015-03-07:
@amoboy 

ss-rules 目前 -e 参数对应的 uci 就是 ipt_ext.
只允许常用的几个端口可以这样:
`uci set shadowsocks.@shadowsocks[-1].ipt_ext='-m multiport --dports 22,80,443'`
或者指定一个范围:
`uci set shadowsocks.@shadowsocks[-1].ipt_ext='--dport 21:5000'`
最后提交修改:
`uci commit shadowsocks`

具体请看 Wiki

kukki, 2015-05-28:
ss-redir 没有这个参数项, 请问怎样实例同样效果?

houzi-, 2015-06-27:
@aa65535 您好！
请问我是在tomato上部署的shadowsocks-libev+ChinaDNS+chnroutes
脚本如下:
```shellscript
#!/bin/sh
# Copyright (C) 2007-2012 OpenWrt.org

iptables -t nat -N SHADOWSOCKS
iptables -t nat -A SHADOWSOCKS -d 0.0.0.0/8 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 10.0.0.0/8 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 127.0.0.0/8 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 169.254.0.0/16 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 172.16.0.0/12 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 192.168.0.0/16 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 224.0.0.0/4 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 240.0.0.0/4 -j RETURN
iptables -t nat -A SHADOWSOCKS -d 157.7.109.114 -j RETURN  
iptables -t nat -A SHADOWSOCKS -p tcp -m set --set gfwlist dst -j RETURN
iptables -t nat -A SHADOWSOCKS -p tcp -j REDIRECT --to-ports 1080
iptables -t nat -I PREROUTING -p tcp -j SHADOWSOCKS
```

```shellscript
#!/bin/sh
# Copyright (C) 2015 OpenWrt.org

_input=/jffs/chnroute.txt
IPS=/usr/sbin/ipset

ipset -N gfwlist nethash

egrep -v "^#|^$" $_input | while IFS= read -r ip

do
        $IPS -A gfwlist $ip	
done
```
我该如何指定ss-redir走特定的端口，因为BT的时候老是跑流量！

期待你的回答，谢谢！

aa65535, 2015-06-27:
@houzi- 
在 `iptables -t nat -I PREROUTING -p tcp -j SHADOWSOCKS` 这条命令中间添加
`-m multiport --dports 22,80,443` 或者 `--dport 21:5000` 即可。
如 `iptables -t nat -I PREROUTING -p tcp --dport 21:5000 -j SHADOWSOCKS`

houzi-, 2015-06-27:
@aa65535 非常感谢您的指导！

flipphos, 2015-06-27:
@aa65535 搭车求指导

我在iptables中最后加入类似规则，但是并没有起到作用，BT还是走了ss流量，iftop能看到发往SS-server的都是高端口（>15000）的流量 。求指导。 软件配置是shadowsocks-libev-spec + ChinaDNS + chnroutes. 是不是因为-spec版本的缘故? 

ipset -N gfwlist iphash 
iptables -t nat -A PREROUTING -p tcp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080 
iptables -t nat -A OUTPUT -p tcp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080

#try this to exclude the torrent traffic
iptables -t nat -I PREROUTING -p tcp --dport 21:5000 -j gfwlist

aa65535, 2015-06-27:
如果你要查看BT走哪些端口，可以关掉shadowsocks 后再使用iftop。
只有spec版带了这个功能，具体看上面提示。
如果要求不高，可以使用 -m multiport --dports 22,80,443 这个格式只指定几个特定端口。

---
https://github.com/shadowsocks/openwrt-shadowsocks/issues/36#issuecomment-116049056

flipphos, 2015-06-27:
因为我专用一台电脑（x.x.x.102）开BT，所以只需用iftop过滤.102的流量即可很方便地查看，看到的流量端口都是>15000.
我的问题是：我在/etc/firewall.user中加入

iptables -t nat -I PREROUTING -p tcp --dport 21:5000 -j gfwlist

或者
iptables -t nat -I PREROUTING -p tcp -m multiport --dports 22,80,443 -j gfwlist

均无法达到排除BT流量走ss通道的目的。（更改后均重启路由，路由为OPENWRT x86） 软件配置：shadowsocks-libev-spec + ChinaDNS + chnroutes.

aa65535, 2015-06-27:
既然是专用的机器用来BT，你应该使用LuCI上提供的访问控制功能，在LAN标签下选择仅允许列表外，然后添加BT机器的IP即可。
2015年6月27日 22:11于 flipphos <notifications@github.com>写道：
> 因为我专用一台电脑（x.x.x.102）开BT，所以只需用iftop过滤.102的流量即可很方便地查看，看到的流量端口都是>15000.
> 我的问题是：我在/etc/firewall.user中加入
>
> iptables -t nat -I PREROUTING -p tcp --dport 21:5000 -j gfwlist
>
> 或者
> iptables -t nat -I PREROUTING [...]

flipphos, 2015-06-27:
访问控制确实有效，之前我用iptables排除单独的内网ip也同样有效。
只不过看到houzi同学的提问诱发了我的强迫症，觉得能够通过控制端口范围来走ss也是挺实用的。特来求指点。

cjh-github, 2015-06-27:
我是路由器运行 transmission，没办法排除单独IP，
但是直接改的 /etc/init.d/shadowsocks
	/usr/bin/ss-rules \
		-c "$CONFIG_FILE" \
		-i "$ignore_list" \
		-a "$ac_args" \
		-b "$wan_bp_ip" \
		-w "$wan_fw_ip" \
		-e "--dport 0:1024 " \
		-o
没问题~~流量没偷跑

flipphos, 2015-06-27:
安装的是shadowsocks-libev 还是libev-spec？哪个版本？谢谢！

cjh-github, 2015-06-27:
github 回复怎么引用啊？？

shadowsocks-libev-spec - 2.1.4
不过我没用 ipset

flipphos, 2015-06-27:
spec版本无需ipset了，国外IP自动走ss的。

flipphos, 2015-06-27:
你的方法有效，非常感谢分享！
我按照你的设置添加参数后重启ss服务，在路由端和ss-server端都能看到没有从ss-server偷跑流量。

flipphos, 2015-06-27:
我的ss版本
shadowsocks-libev-spec	2.2.1-1
        /usr/bin/ss-rules \
                -c "$CONFIG_FILE" \
                -i "$ignore_list" \
                -a "$ac_args" \
                -b "$wan_bp_ip" \
                -w "$wan_fw_ip" \
                -e "$ipt_ext" \
                -e "--dport 0:1024" \
                -o $udp

希望后续版本能提供端口范围设置就更方便了。

sjtuross, 2015-12-01:
通过以下命令貌似可以阻止指定端口走SS，比如BT之类的，要是ipt_ext有对应的luci字段来设置就更方便了
```
uci set shadowsocks.@shadowsocks[-1].ipt_ext='-m multiport ! --dports 23233,51547'
uci commit shadowsocks
/etc/init.d/shadowsocks restart
```

aa65535, 2015-12-03:
@sjtuross 这个功能不常用,而且对于不熟悉的用户来说容易出现问题,所以只能使用命令行或者编辑 /etc/config 下的文件来修改

zgc1209, 2016-02-06:
@aa65535 
您好，在openwrt15.05上，shadowsocks-spec 2.4.4版的参数是这样的：
/usr/bin/ss-rules \
		-s "$server" \
		-l "$local_port" \
		-S "$udp_server" \
		-L "$udp_local_port" \
		-a "$ac_ips" \
		-i "$(uci_get_by_type access_control wan_bp_list)" \
		-b "$(uci_get_by_type access_control wan_bp_ips)" \
		-w "$(uci_get_by_type access_control wan_fw_ips)" \
		-e "$(uci_get_by_type access_control ipt_ext)" \
		-o $ARG_UDP
	return $?

运行uci set shadowsocks.@shadowsocks[-1].ipt_ext='-m multiport --dports 22,80,443'
提示uci：Invalid argument

请问需要怎么修改才可以指定端口，谢谢。

aa65535, 2016-02-06:
将 @shadowsocks 改成@access_control ，因为配置文件机构改变了。

https://github.com/shadowsocks/openwrt-shadowsocks/issues/36#issuecomment-180716790

nanxiaobei, 2016-03-20:
@aa65535 

请问一下运行 uci set shadowsocks.@access_control[-1].ipt_ext='-m multiport --dports 22,80,443' 后，

添加的 '-m multiport --dports 22,80,443' 在哪个文件里？

比如我想删除这个规则或者修改它，

用 WinSCP 打开 OpenWrt 应该在哪个文件里修改呢？

aa65535, 2016-03-20:
/etc/ config/ shadowsocks
@aa65535

请问一下运行 uci set shadowsocks.@access_control[-1].ipt_ext='-m multiport --dports 22,80,443' 后，

添加的 '-m multiport --dports 22,80,443' 在哪个文件里？

比如我想删除这个规则或者修改它，

用 WinSCP 打开 OpenWrt 应该在哪个文件里修改呢？

---
https://github.com/shadowsocks/openwrt-shadowsocks/issues/36#issuecomment-198986895

cokebar, 2016-03-21:
@aa65535 我觉得完全可以在luci中增加针对端口的ACCESS CONTROL

nanxiaobei, 2016-03-22:
@flipphos 你好，麻烦问下，如何使用 iftop 查看 bt 下载是否走了 shadowsocks 代理？

flipphos, 2016-03-23:
@nanxiaobei 我用笨办法验证，在pt/bt下载并且下载速度较大的时候（比如2MB/s），ssh连入vps，运行iftop，这时能看到vps上的进站流量速度和出站到你自己的国内ip的速度，和你的pt速度相差很大。另外也可在iftop中验证pt下载的peer ip是否经过ss。 设置正确的话，pt是不走ss流量的。我经过1年用下来可以确定这点。

muronghan, 2016-06-20:
安装的是2.4.7版本的shadowsocks-libev-spec在openwrt路由上，如果我不想代理某端口，比如51413，应该手动在/etc/config/shadowsocks里添加什么呢？是否在文件的config access_control部分里添加：

option ipt_ext '--dport! 51413'

aa65535, 2016-06-21:
@muronghan `option ipt_ext '! --dport 51413'`

muronghan, 2016-06-21:
@aa65535 很感谢你的回答（以及所有的努力）。。。按你给的命令行试了一下，好像不对。方法是改/etc/config/shadowsocks文件，重新启动shadowsocks，然后用NAS上的transmission服务下载一个视频文件（大多来自国外，我开的是白名单，所以如果不停掉51413端口，就会走ss流量），然后看我vps的流量图，图示流量一下剧增，所以应该是走了流量。

作为对比试验，如果设为一个区段，比如option ipt_ext '--dport 21:5000’，则没有走ss流量，故此认为这个命令行没起作用。

注：将指向路由51413端口的数据port forward到NAS的51413端口。

aa65535, 2016-06-21:
51413只是连接tracker服务器所使用的端口，实际下载使用的是随机端口，所以不会有效。

muronghan, 2016-06-22:
@aa65535 明白了，非常感谢！如果是这样，把要代理的端口设在低区，比如option ipt_ext '--dport 0:1023’会比较好，因为随机使用的端口一般也在高区。这样避免了下载时跑ss流量，从而遭致VPS服务商的警告信。

题外话，其实本来是打算通过netflix看付费的视频，但netflix目前还不支持中国，如果用ss代理，则netflix会自动识别，并停止流媒体的播放，实在是逼良为娼，呵呵。

FONGto, 2016-08-07:
```
	/usr/bin/ss-rules \
		-c "$CONFIG_FILE" \
		-i "$ignore_list" \
		-a "$ac_args" \
		-b "$wan_bp_ip" \
		-w "$wan_fw_ip" \
		-e "$ipt_ext" \
		_-e "--dport 21:1024" \_
		-o $udp
	return $?
```
实测这个可以设置指定代理端口

haruhitamako, 2017-01-06:
使用option ipt_ext '--dport 0:1023’设置指定端口使用ss流量后会和chnroute表冲突。导致指定端口的所有流量都走了ss。包括国内ip。

cokebar, 2017-01-06:
@haruhitamako 按理说不会，根据代码来看，首先会对目标IP进行匹配 chnroute中的IP在这一条中就已经被RETURN了。
`	-A SS_SPEC_LAN_DG -m set --match-set ss_spec_dst_sp dst -j RETURN
	-A SS_SPEC_LAN_DG -p $protocol $EXT_ARGS -j SS_SPEC_LAN_AC`

haruhitamako, 2017-01-06:
@cokebar 实际上刚设置好重启ss是正常的，但使用过程中就会自动转为全局。本来以为是chinadns的问题，但重启chinadns没效果。重启ss或者重启路由器就会自动使用chnroute，但是很快又变全局了。

cokebar, 2017-01-06:
@haruhitamako 使用过程中转为全局 应该是其他程序触发的

muronghan, 2017-01-07:
这个参数，我这里使用也没有什么问题。