shadowsocks/openwrt-shadowsocks #39: Bug report for brcm47xx serial routers

hduffddybz, 2015-01-16:
Recently, I port the openwrt to the ASUS RT-N16 router, which is brcm 47** serial. I try to use the shadowsocks client for openwrt(which version is shadowsocks_spec_2.0.7.ipk downloaded from sourceforge),I try to use the shadowsocks as the global proxy for the router and also use CHINADNS plus shadowsocks to cross the great wall, but almost all the time,it can only access to google and gmail, some websites such as facebook & twitter which is https version also blocked.Many people encounter this problem, any methods to solve it? In addition,It seems that Windows client did not have this problem.

hduffddybz, 2015-01-16:
Please read the comments on this site 
http://cokebar.info/archives/664
many people have problems that cannot access to facebook,twitter and dropbox

aa65535, 2015-01-16:
I can't reproduce the issue.
Maybe you can put up the configuration of DNS.
You can also use the dig command to test it.

hduffddybz, 2015-01-17:
method1(It can only access to google and Gmail, cannot access to facebook ,twitter and dropbox):
[image]
I try to use the dig@192.168.1.1 www.google.com.hk & dig @8.8.8.8 www.google.com.hk and the same method to youtube, It get the same ip address, but only the facebook can access. The same method applied to facebook, but the router get the wrong ip address, so I confirm that existing DNS Pollution. So I try the method 2:

[image]
[image]
[image]
[image]
sometimes it can access to google and gmail, sometimes all the websites from foreign countried are all blocked. When using dig@192.168.1.1 www.google.com.hk, always get the connection timeout. But the dig@8.8.8.8 www.google.com.hk all done well. Any bug existed in ChinaDNS & shadowsocks-libev-spec?

aa65535, 2015-01-17:
Try this:
 - Fill in the `127.0.0.1#5353` to `Local server` instead of `DNS forwardings`.  
 - Execute the following command, and check the verbose logs.  
   `/usr/bin/chinadns -s 114.114.114.114,127.0.0.1:5151 -p 5353 -c /etc/chinadns_chnroute.txt -l /etc/chinadns_iplist.txt -y 0.3 -v`

`Local server` same with` DNS forwardings`, but you need to let them have a null.
Maybe you can use other faster the DNS server.
The v2.0.8 has been released, update it.

hduffddybz, 2015-01-17:
- First, I stop the luci-app-chinadns and in the ssh command , command /usr/bin/chinadns -s 114.114.114.114,127.0.0.1:5151 -p 5353 -c /etc/chinadns_chnroute.txt -l /etc/chinadns_iplist.txt -y 0.3 -v

- Second, I type dig@192.168.1.1 -p 5353 www.youtube.com, it returns 
Sat Jan 17 05:31:10 2015 request www.youtube.com
Sat Jan 17 05:31:10 2015 response www.youtube.com from 114.114.114.114:53 - 78.47.140.88, filter
but the 78.47.140.88 is wrong ip address(in the chrome Browser, it access to the wrong website), and the dns query did not pass to shadowsocks udp port 5151.

and also type dig@8.8.8.8 www.youtube.com, it returns 37.61.54.158 ip address. I type it in the chrome Browser, but cannot access it. It seems that this ip address cannot go through from shadowsocks client to shadowsocks server?

- Last, same problems have taken place in the query of Facebook.

aa65535, 2015-01-17:
`78.47.140.88` and `78.47.140.88` should be fake ip.
No answer from the `127.0.0.1:5151`?
Try `dig@192.168.1.1 -p 5151 www.youtube.com` check it.

hduffddybz, 2015-01-17:
Sorry, It does not work.

aa65535, 2015-01-17:
Have you enable udprelay mode on shadowsocks remote server?

hduffddybz, 2015-01-17:
But the problem is that even I get the correct IP address, I also cannot connect it.

aa65535, 2015-01-17:
I think you should change the server in order to exclude the issue of the service side.

hduffddybz, 2015-01-19:
Sorry,still cannot slove this problem, the command "dig @192.168.1.1 -p 5151 www.youtube,com" cannot resolve the domain name  even the server use the parameter "-u" to enable udprelay mode.The result are below:
[image]

aa65535, 2015-01-19:
This is DNS poisoning.
If udprelay mode of shadowsocks not work correctly, you can try to build 
a dns server on the server side. Reference to [Plan3](https://sourceforge.net/p/openwrt-dist/wiki/Plan3/) expanded section.
BTW, The v2.1.0 has been released.

Good luck!

hduffddybz, 2015-01-19:
Finally, We find that if only use the udprelay, all things do well, but when when using chinadns+udprely, almost all the domains are submited to 114.114.114.114. In addition,shadowsocks can distinguish the ip address from home or aboard?

aa65535, 2015-01-19:
You have modified the `/etc/chinadns_chnroute.txt`?
ChinaDNS according to it to distinguish the location of the server.
You should check the verbose log of ChinaDNS when necessary.

Shadowsocks-libev-spec has a built-in CHNRoute is 
`/etc/shadowsocks/ignore.list`, but only for TCP traffic.

Postscript:  
```
$ dig @192.168.1.254 -p5353 www.youtube.com

; <<>> DiG 9.8.4-rpz2+rl005.12-P1 <<>> @192.168.1.254 -p5353 www.youtube.com
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 23404
;; flags: qr rd ra; QUERY: 1, ANSWER: 5, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;www.youtube.com.		IN	A

;; ANSWER SECTION:
www.youtube.com.	21369	IN	CNAME	youtube-ui.l.google.com.
youtube-ui.l.google.com. 69	IN	A	173.194.72.91
youtube-ui.l.google.com. 69	IN	A	173.194.72.190
youtube-ui.l.google.com. 69	IN	A	173.194.72.136
youtube-ui.l.google.com. 69	IN	A	173.194.72.93

;; Query time: 96 msec
;; SERVER: 192.168.1.254#5353(192.168.1.254)
;; WHEN: Mon Jan 19 12:11:33 2015
;; MSG SIZE  rcvd: 131
```
```
# /usr/bin/chinadns -s 114.114.114.114,127.0.0.1:5151 -p 5353 -c /etc/chinadns_chnroute.txt -l /etc/chinadns_iplist.txt -y 0.3 -v
ChinaDNS 1.2.2
Mon Jan 19 12:08:32 2015 request www.youtube.com
Mon Jan 19 12:08:32 2015 response www.youtube.com from 114.114.114.114:53 - 208.93.0.150, filter
Mon Jan 19 12:08:32 2015 response www.youtube.com from 127.0.0.1:5354 - 173.194.72.93, 173.194.72.136, 173.194.72.190, 173.194.72.91, pass
```

hduffddybz, 2015-01-19:
Thank you for your reply, I try the chinadns + shadowsocks again and again, finally I find that the access to facebook & youtube is not stable because of the problem of dns resolving(the shadowsocks udprely).Any plans to review the code?

aa65535, 2015-01-19:
If udprely doesn't work correctly, ChinaDNS should return connection timeout rather than fake IP.
It's not a problem of shadowsocks-libev. I think you should check elsewhere.

hduffddybz, 2015-01-20:
Thanks. I try to use the dnsmasq on the remote server, and now all things do well.

closed 2015-01-20 by hduffddybz