shadowsocks/openwrt-shadowsocks #51: 编译错误：No targets specified and no makefile found.  Stop.

EdiTurn, 2015-03-14:
最近更新了一下 OpenWRT trunk（WNDR4300，ar71xx/nand），发现由于最近的 Openwrt 开始采用 glibc，所以提供的预编译 ipk 无法使用，故自行编译了一下。

<pre><code>root@XXXXXX:/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64# make package/shadowsocks-libev/compile V=99
#
#configuration written to .config
#
make[1]: Entering directory `/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64'
make[2]: Entering directory `/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/package/shadowsocks-libev'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float  -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/target-mips_34kc_glibc-2.19/usr/include -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/target-mips_34kc_glibc-2.19/include -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_glibc-2.19/usr/include -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_glibc-2.19/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float  -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/target-mips_34kc_glibc-2.19/usr/include -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/target-mips_34kc_glibc-2.19/include -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_glibc-2.19/usr/include -I/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_glibc-2.19/include " LDFLAGS="-L/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/target-mips_34kc_glibc-2.19/usr/lib -L/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/target-mips_34kc_glibc-2.19/lib -L/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_glibc-2.19/usr/lib -L/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_glibc-2.19/lib " make   -C /tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/build_dir/target-mips_34kc_glibc-2.19/shadowsocks-libev/openssl/shadowsocks-libev-2.1.4/. AR="mips-openwrt-linux-gnu-gcc-ar" AS="mips-openwrt-linux-gnu-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float" LD=mips-openwrt-linux-gnu-ld NM="mips-openwrt-linux-gnu-gcc-nm" CC="mips-openwrt-linux-gnu-gcc" GCC="mips-openwrt-linux-gnu-gcc" CXX="mips-openwrt-linux-gnu-g++" RANLIB="mips-openwrt-linux-gnu-gcc-ranlib" STRIP=mips-openwrt-linux-gnu-strip OBJCOPY=mips-openwrt-linux-gnu-objcopy OBJDUMP=mips-openwrt-linux-gnu-objdump SIZE=mips-openwrt-linux-gnu-size CROSS="mips-openwrt-linux-gnu-" ARCH="mips" ;
make[3]: Entering directory `/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/build_dir/target-mips_34kc_glibc-2.19/shadowsocks-libev/openssl/shadowsocks-libev-2.1.4'
make[3]: *** No targets specified and no makefile found.  Stop.
make[3]: Leaving directory `/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/build_dir/target-mips_34kc_glibc-2.19/shadowsocks-libev/openssl/shadowsocks-libev-2.1.4'
make[2]: *** [/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/build_dir/target-mips_34kc_glibc-2.19/shadowsocks-libev/openssl/shadowsocks-libev-2.1.4/.built] Error 2
make[2]: Leaving directory `/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64/package/shadowsocks-libev'
make[1]: *** [package/shadowsocks-libev/compile] Error 2
make[1]: Leaving directory `/tmp/OpenWrt-SDK-ar71xx-nand_gcc-4.8-linaro_glibc-2.19.Linux-x86_64'
make: *** [package/shadowsocks-libev/compile] Error 2
</pre></code>

而后尝试编译 shadowsocks/shadowsocks-libev 倒是成功。

EdiTurn, 2015-03-14:
顺便问一下，以后是否能够同时编译一下 glibc 的预编译版本，谢谢。

aa65535, 2015-03-14:
使用 generic 目录下的 SDK 不可以？

EdiTurn, 2015-03-14:
没有尝试 generic 目录下的 SDK，因为 generic 还是用 uClibc 的……

EdiTurn, 2015-03-14:
用 generic 的 SDK 试了一下，同样的错误信息。

myliyifei, 2015-03-14:
sdk从来没有成功过

wongsyrone, 2015-03-15:
找到一个说明，不知道是否和toolchain的并行编译有关，有条件的可以自己从源码编译一遍
如果使用的是glibc，则在编译toolchain阶段不能使用make -j num多核/并行编译。在toolchain已经编译完成的情况下，其他target，packages阶段的都可多核/并行编译！
原因：由于glibc-ports和glibc之间并没有在Makefile中定义DEPENDS依赖关系（glibc-ports依赖于glibc），因此在make -j num使用多核/并行编译时，可能会glibc-ports先于glibc/gcc/binutil编译，导致编译出错

aa65535, 2015-03-15:
这个是 SDK 出了问题导致 autoreconf 不能正常完成。

shadowsocks/shadowsocks-libev 可以成功编译是因为源码中自带了 `configure` 和 `auto/` ，即使 autoreconf 出错也可以继续执行下一步。

错误：
```
bash: /home/rin/openwrt/wndr4300/SDK/staging_dir/host/bin/autoreconf: /build/openwrt/slave/ar71xx.nand/build/staging_dir/host/bin/perl: bad interpreter: No such file or directory

/home/rin/openwrt/wndr4300/SDK/staging_dir/host/bin/aclocal: 2: /home/rin/openwrt/wndr4300/SDK/staging_dir/host/bin/aclocal: aclocal.real: not found
autoreconf: /home/rin/openwrt/wndr4300/SDK/staging_dir/host/bin/aclocal failed with exit status: 127
```

建议使用 barrier_breaker 目录下的 SDK。

closed 2015-03-15 by aa65535

LazyZhu, 2015-03-15:
OpenWRT trunk 又要大折腾改成 eglibc了...

GreatFireWall, 2016-04-01:
用 [stable SDK OpenWrt 'Barrier Breaker' x86-64 • brcm47xx](https://downloads.openwrt.org/barrier_breaker/14.07/brcm47xx/generic/OpenWrt-SDK-brcm47xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2.tar.bz2) 还是报错.

> 
make[3]: Entering directory '/home/ec2-user/OpenWrt-SDK-brcm47xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5'
make[3]: *** No targets specified and no makefile found.  Stop.
make[3]: Leaving directory '/home/ec2-user/OpenWrt-SDK-brcm47xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5'
make[2]: *** [/home/ec2-user/OpenWrt-SDK-brcm47xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.4.5/.built] Error 2
make[2]: Leaving directory '/home/ec2-user/OpenWrt-SDK-brcm47xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/package/shadowsocks-libev'
make[1]: *** [package/shadowsocks-libev/compile] Error 2
make[1]: Leaving directory '/home/ec2-user/OpenWrt-SDK-brcm47xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2'
make: *** [package/shadowsocks-libev/compile] Error 2