shadowsocks/openwrt-shadowsocks #52: 请问如何解决中间人攻击

lixingcong, 2015-03-20:
用的是ar71xx版的shadowsocks+ChinaDNS，近期发现个别网站ssl证书错误，如何解决？

[image]

aa65535, 2015-03-21:
先解决 DNS 污染。

closed 2015-03-21 by aa65535

wogong, 2015-05-10:
@aa65535 您好，首先感谢您的工作。路由层面的智能代理之前一直没有力气折腾，因为VPN的不稳定及复杂性导致屡次放弃，这次利用ss方案非常方便地解决了这个问题。下面有一个在使用阶段遇到的小问题。

我在路由同时使用 chinadns+dnsmasq+openwrt-shadowsocks，之前我想着已经利用前二者解决了DNS污染问题，故取消了openwrt_shadowsocks中的UDP转发，但是还是遇到了本issue的问题，打开了这个选项就没有问题了。故冒昧请问openwrt_shadowsocks中的UDP转发具体是指什么，是否一定要打开（或者是什么情况下不需要打开）？多谢。

aa65535, 2015-05-11:
@wogong 
UDP转发就是使用 ss-tunnel 提供给本地 DNS 查询使用。
如果单使用 chinadns+dnsmasq 的话，打开chinadns的压缩指针。

wogong, 2015-05-11:
@aa65535 多谢，很久没关注，没注意到到有了指针压缩的功能，看了issue才大概了解。希望现在的方案可以稳定。再次感谢：）