shadowsocks/openwrt-shadowsocks #54: 请问shadowsocks是不是和nwan的多拨有冲突，多拨的情况下无法工作

tank8888jx, 2015-03-23:
本人使用nwan进行多拨情况下无法使用shadowsocks，shadowsocks采用的是忽略国内ip段方式工作，是否是shadowsocks无法判断多拨情况的外网出口而无法工作，请教应该如何修改设置。

billysun76, 2015-03-24:
应该没有问题，我就是在多拨下用的ss，你应该先检查一下ss服务器是否正常连接。

aa65535, 2015-03-24:
使用 `iptables-save` 命令将规则导出看看。

flipphos, 2015-03-24:
我一直有这个疑问，在OpenWRT下安装了dnsmasq-full, Shadowsocks-spec和ChinaDNS，ChinaDNS加载了ignore.list，dnsmasq加载了[黑名单.list]，我用ipset定义了黑名单和ip规则，在黑名单中的网站直接通过ss-redir转发了，但是发现实际上ss-redir转发了所有的国外网站了，这是我哪里设置不对吗？

/etc/firewall.user中规则如下，ss-redir监听1080端口。
ipset -N gfwlist iphash
iptables -t nat -A PREROUTING -p tcp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080
iptables -t nat -A OUTPUT -p tcp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080

如果转发规则生效的话，我访问www.formyip.com <http://www.formyip.com/>应该显示我国内IP，而不是ss-server的IP。但是现在访问显示的是ss-server的IP.

求解惑，谢谢！

aa65535, 2015-03-24:
@flipphos 如果你需要按域名的代理，就不要使用 spec 版的 shadowsocks，可以参考 https://cokebar.info/archives/962 设置。

flipphos, 2015-03-24:
多谢指点，弄好了。之前一直没搞清楚spec版本和非spec版本的区别。原来spec版本就是根据国内外IP分流，而非spec版本可以结合ipset按域名代理。希望在项目的主页上或者wiki上能够解释一下（或许是我眼拙没看到）。估计像我这样迷惑的人也会不少。我之前在不同地方问过，一直没弄懂。

此外，我在路由上再运行一个ss-local，监听另外一个端口1090 (ss-redir是监听1080端口),指向另外一个ss-server，以便第一个ss-server不可用的时候可以在浏览器中手动指定192.168.1.1:1090上网（或者设置foxyproxy之类)。这样运行可以吗？

/usr/bin/ss-local -c /etc/shadowsocks/config2.json -b 0.0.0.0  -l 1090 -f /var/run/shadowsocks2.pid

ss-redir未来版本能够支持配置多个ss-server进行冗余就好了，像COW一样，一个服务器不通就自动连接下一个。

xlcwzx, 2015-03-24:
Hi flipphos

我猜你可能对这篇基于haproxy进行多服务器组合的官方wiki感兴趣。
https://github.com/shadowsocks/shadowsocks/wiki/Setup-a-Shadowsocks-relay
当然，这个适合在本地的Linux服务器上或者额外的VPS上部署（比如我部署在了阿里云上，不过一直没在用）。

另外，自动状态检查和自动重播在这里是有参考脚本的：https://cokebar.info/archives/850#part3
我建议你看看，修改一下可能就可以实现你的服务器切换的目的。

flipphos, 2015-03-24:
哈哈，看到haproxy这个词，不禁感叹茫茫人海，你真是知我！

我之前就是挂念着单机上COW的好用，想着在OPENWRT能实现就好了，到处搜，结果就搜到了Haproxy。一查，果然有BB(14.09)下能用的版本，装上之后捣鼓了一下，没成功，结果草草收场。

 

偏个题我讨教一下：

我现在黑名单上的网站访问是扔给5353端口的ss-redir的，假设我停掉ss-redir，让haproxy接管5353端口，按照这篇wiki，这样配置，可行吗？

---------------------------------------------

frontend ss-in

        bind *:5353

        default_backend ss-out

 

backend ss-out

        server server1 US_VPS_IP:8388 maxconn 20480

        server server2 JP_VPS_IP:8388 maxconn 20480

       server server3 UK_VPS_IP:8388 maxconn 20480

---------------------

 

我还想在Openwrt上试试看。

 

Cokebar的参考脚本我之前看过，你的提醒很有道理，改写一下果然有更多用处。

 

https://github.com/shadowsocks/shadowsocks/wiki/Setup-a-Shadowsocks-relay

xlcwzx, 2015-03-25:
不行，你应该再开启一个ss-redir，服务器填写OpenWRT的IP和haproxy
frontend端口，然后把你的访问流量导到这个新的ss-redir的监听端口上去。注意这要求远端服务器共用同样的加密方式和用户名密码。

另外这个issue是不是该close了？我也是刚开始捣鼓，有想要讨论的欢迎给我发邮件和私信。

flipphos, 2015-03-25:
我昨晚套用wiki里面的配置文件捣鼓了一下，我感觉原理上应该对了，但是还没成功，ss-redir和ss-tunnel都试了，正想找你详聊一下。如何发邮件给你，或者私信？看不到你的邮箱啊。

closed 2015-04-30 by aa65535