shadowsocks/openwrt-shadowsocks #57: 使用SDK编译报错，缺少依赖libcrypto.so.1.0.0

bcheng049, 2015-04-14:
Package shadowsocks-libev is missing dependencies for the following libraries:
libcrypto.so.1.0.0
SDK使用的自己编译的版本，OpenWrt-SDK-ar71xx-generic_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64
我把代码放到编译openwrt的package中可以编译成功，单独使用SDK来编译就会缺少依赖，请问下是什么原因呢？

wongsyrone, 2015-04-15:
可能openssl没有创建符号链接？

bcheng049, 2015-04-15:
我在SDK的目录查找了下，在SDK的目录./staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib/libcrypto.so.1.0.0下找到了这个库。
我在编译openwrt的项目下面找到多个地方有这个库
./build_dir/target-mips_34kc_uClibc-0.9.33.2/openssl-1.0.2a/ipkg-ar71xx/libopenssl/usr/lib/libcrypto.so.1.0.0
./build_dir/target-mips_34kc_uClibc-0.9.33.2/openssl-1.0.2a/libcrypto.so.1.0.0
./build_dir/target-mips_34kc_uClibc-0.9.33.2/openssl-1.0.2a/ipkg-install/usr/lib/libcrypto.so.1.0.0
./staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib/libcrypto.so.1.0.0
./staging_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx/usr/lib/libcrypto.so.1.0.0
但是直接把SDK解压之后编译就缺少依赖，不知何解

closed 2015-04-18 by aa65535