shadowsocks/openwrt-shadowsocks #59: 接口重连后zone_lan_prerouting链里面规则丢失，求修复

Sakura-Winkey, 2015-04-20:
大触你好，我又来烦你了（够！

因为诸多蛋疼的原因，我家路由在wan zone接口有三个（管理光猫的moden，外网wan，6in4隧道wan6）
发现如果单独重连wan6（6in4隧道偶尔不稳定要重启）那么防火墙里面

zone_lan_prerouting链

会丢失

SS_SPEC_LAN_AC 选项为/* ss_spec_rules*/（这说明太蛋疼了

造成路由本身能代理但『路由下面的设备不能无痛代理』
求修正，渣渣在此谢过

aa65535, 2015-04-25:
这个问题暂时无法验证，你可以先试试将 ss-rules 里的 `zone_lan_prerouting` 替换成 `PREROUTING`。

Sakura-Winkey, 2015-04-25:
要验证很简单的，我猜测maybe重连哪个接口都会出现这样的情况（防火墙搜索ss_spec会发现比原来少了两条，就是在nat表的zone_lan_prerouting链里面
稍后我试一下添加到PREROUTING，谢谢
另外，是否可以考虑修改下makefile，让shadowsocks-libev-spec依赖ipset和ip这两个包，然后把iptable规则写到（比如/var/shadowsocks.user）文件里面，并且在防火墙的自定义规则里加上

[[ -s /var/shadowsocks.user]] && . /var/shadowsocks.user

这样，并且改写下『init的rules()函数』或者是『rules本身』，让他
1.每次执行前先删除旧有的shadowsocks.user文件
2.如果执行了stop就删除shadowsocks.user，并且重启防火墙

另外shadowsocks的luci偶尔发现点了保存应用后不是马上生效，所以我自己加了cbi.apply动作（shadowsocks restart），希望对你有帮助

aa65535, 2015-04-26:
如果 `zone_lan_prerouting` 会丢失，应该属于 Openwrt 的 BUG，使用 `PREROUTING` 应该避免这个问题。

至于依赖关系，shadowsocks-libev-spec 并没有用到 ip，另外 ipset 也不是必须的，因为有些 Openwrt 版本是无法安装 ipset 的，添加依赖后也会导致安装出问题。

iptable 规则是动态生成的，不需要临时文件去存储，而且在 stop() 后重启防火墙并不是一个好主意。

LuCI 点击 `保存&应用` 无效是因为启动项里的自动启动没打开。

commit 3e22783d (2015-05-07): fixed #59
closed 2015-05-07 by aa65535