shadowsocks/openwrt-shadowsocks #64: Help to test TPROXY

madeye, 2015-05-04:


aa65535, 2015-05-04:
Good job. I'll test it.

aa65535, 2015-05-04:
recipe for target 'ss_redir-udprelay.o' failed
```
udprelay.c: In function 'get_dstaddr':
udprelay.c:148:57: error: 'IP_RECVORIGDSTADDR' undeclared (first use in this function)
         if (cmsg->cmsg_level == SOL_IP && cmsg->cmsg_type == IP_RECVORIGDSTADDR) {
                                                         ^
udprelay.c:148:57: note: each undeclared identifier is reported only once for each function it appears in
udprelay.c: In function 'remote_recv_cb':
udprelay.c:650:33: error: 'IP_TRANSPARENT' undeclared (first use in this function)
     if (setsockopt(src_sock, SOL_IP, IP_TRANSPARENT, &opt, sizeof(opt))) {
                                 ^
```

may be required:
```c
#ifndef IP_TRANSPARENT
#define IP_TRANSPARENT       19
#endif

#ifndef IP_RECVORIGDSTADDR
#define IP_RECVORIGDSTADDR   20
#endif
```

madeye, 2015-05-05:
Tests passed on OpenWRT 14.07. Merged to the master branch.

closed 2015-05-05 by madeye

shifengjiang, 2015-12-21:
have you tested on Linux router? I tried to setup on my debian 8.2 but found failed. On my another old Debian 7.x box with shadowsocks-libev 2.0.1 it works. B.T.W. I noticed the ss-dir listening address changed from 0.0.0.0 to 127.0.0.1. Does it matter?