shadowsocks/openwrt-shadowsocks #73: PandoraBox 无法安装 shadowsocks-libev-spec 提示内核依赖有问题

lvii, 2015-05-10:
路由器 | 小米 mini
-----  | --------------
固件版本  | PandoraBox 14.09 r512 (20150309) stable

**已删除** PandoraBox 默认自带的 `shadowsocks-libev` 和 `luci-app-shadowsocks`

尝试安装 sourceforge.net openwrt-dist 中的 `shadowsocks-libev-spec_2.2.0-1_ramips_24kec.ipk`

http://sourceforge.net/projects/openwrt-dist/files/shadowsocks-libev/2.2.0-ff56476/ramips/

提示 kernel 依赖报错 `shadowsocks-libev-spec` **为什么会依赖内核呢？**

```
[root@PandoraBox:/root]#opkg install shadowsocks-libev-spec
Installing shadowsocks-libev-spec (2.2.0-1) to root...
Downloading http://openwrt-dist.kxswq.ml/releases/ramips/packages/shadowsocks-libev-spec_2.2.0-1_ramips_24kec.ipk.
Multiple packages (kmod-ipt-conntrack and kmod-ipt-conntrack) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipv6 and kmod-ipv6) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ip6tables and kmod-ip6tables) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for shadowsocks-libev-spec:
 *      kernel (= 3.10.49-1-3df3ab26a49a04478a9633bf83827ca3) * 
 * opkg_install_cmd: Cannot install package shadowsocks-libev-spec.
```

强制安装，还是会出现这个报错，不知道这样会不会出现问题

```
[root@PandoraBox:/root]#opkg install shadowsocks-libev-spec --force-depends
Installing shadowsocks-libev-spec (2.2.0-1) to root...
Downloading http://openwrt-dist.kxswq.ml/releases/ramips/packages/shadowsocks-libev-spec_2.2.0-1_ramips_24kec.ipk.
Multiple packages (kmod-ipt-conntrack and kmod-ipt-conntrack) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipv6 and kmod-ipv6) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ip6tables and kmod-ip6tables) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Installing iptables-mod-tproxy (1.4.21-1) to root...
Downloading http://downloads.openwrt.org/barrier_breaker/14.07/ramips/mt7620a/packages/base/iptables-mod-tproxy_1.4.21-1_ramips_24kec.ipk.
Installing kmod-ipt-tproxy (3.10.49-1) to root...
Downloading http://downloads.openwrt.org/barrier_breaker/14.07/ramips/mt7620a/packages/base/kmod-ipt-tproxy_3.10.49-1_ramips_24kec.ipk.
Configuring kmod-ipt-tproxy.
Configuring iptables-mod-tproxy.
Configuring shadowsocks-libev-spec.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for shadowsocks-libev-spec:
 *      kernel (= 3.10.49-1-3df3ab26a49a04478a9633bf83827ca3) * 
```

谢谢！

wongsyrone, 2015-05-11:
iptables扩展涉及到了kmod，也就是内核模块，所以会有内核依赖，这种情况建议自己编译

aa65535, 2015-05-11:
除非 PandoraBox 作者提供全部的内核模块包，否则无解，建议继续使用 v2.1.4

closed 2015-05-11 by aa65535

lvii, 2015-05-11:
@wongsyrone @aa65535 如果是 kmod-* 相关模块包依赖，那应该是 kmod_* 依赖报错。为何会报错shadowsocks-libev-spec 。是否可以将 shadowsocks-libev-spec 简单依赖 kmod-* 相关模块即可。这样每个 rom 都有自己对应版本的 kmod-* 相关模块。谢谢！

wongsyrone, 2015-05-11:
@lvii  你不加强制参数的时候它是不会安装的，正是因为内核版本不满足条件，而加了强制之后就跳过检查相信用户，最后提示了错误让用户注意。
至于你说的”简单依赖kmod-*模块“是不能实现的，这个是linux内核如何插入模块的细节决定的，所以如同aa65535说的“除非 PandoraBox 作者提供全部的内核模块包，否则无解”

lvii, 2015-05-12:
1. @aa65535 不太理解你讲的：**PandoraBox 作者提供全部的内核模块包**  可否展开科普一下
2. @wongsyrone 是哪个软包的依赖导致你说的：**内核版本不满足条件**

查询 shadowsocks-libev-spec 和 iptables  的依赖，ss 没有直接依赖 kmod-* 相关的包：

```
root@OpenWrt:~# opkg info shadowsocks-libev-spec|grep Depends
Depends: libc, libopenssl, resolveip

root@OpenWrt:~# opkg info iptables|grep Depends
Depends: libc, kmod-ipt-core, libip4tc, libip6tc, libxtables
```

查询 kmod-* 模块是和内核版本一致的，都是：3.10.49 版本，这样 iptables 的依赖版本都匹配的

```
root@OpenWrt:~# uname -a
Linux OpenWrt 3.10.49 #3 Wed Oct 1 14:00:51 CEST 2014 mips GNU/Linux

root@OpenWrt:~# opkg files kmod-ipt-conntrack 
Package kmod-ipt-conntrack (3.10.49-1) is installed on root and has the following files:
/lib/modules/3.10.49/xt_state.ko
/lib/modules/3.10.49/iptable_raw.ko
... ...
/lib/modules/3.10.49/nf_conntrack.ko
/lib/modules/3.10.49/xt_conntrack.ko

root@OpenWrt:~# opkg files kmod-ipv6
Package kmod-ipv6 (3.10.49-1) is installed on root and has the following files:
/etc/modules.d/20-ipv6
/lib/modules/3.10.49/ipv6.ko

root@OpenWrt:~# opkg files kmod-ip6tables
Package kmod-ip6tables (3.10.49-1) is installed on root and has the following files:
/etc/modules.d/42-ip6tables
/lib/modules/3.10.49/ip6t_REJECT.ko
/lib/modules/3.10.49/ip6table_mangle.ko
... ...
/lib/modules/3.10.49/ip6table_raw.ko
/lib/modules/3.10.49/nf_conntrack_ipv6.ko

root@OpenWrt:~# opkg files kmod-ipt-core
Package kmod-ipt-core (3.10.49-1) is installed on root and has the following files:
/lib/modules/3.10.49/x_tables.ko
/lib/modules/3.10.49/ip_tables.ko
... ...
/lib/modules/3.10.49/xt_tcpudp.ko
/lib/modules/3.10.49/xt_mark.ko

root@OpenWrt:~# opkg files kmod-ipt-conntrack
Package kmod-ipt-conntrack (3.10.49-1) is installed on root and has the following files:
/lib/modules/3.10.49/xt_state.ko
/lib/modules/3.10.49/iptable_raw.ko
... ...
/lib/modules/3.10.49/nf_conntrack.ko
/lib/modules/3.10.49/xt_conntrack.ko
```

谢谢！

SpecterHi, 2015-10-16:
以来特定内核的是iptables-mod-tproxy
然后这个包是被shadowsocks-libev-spec 2.2.0之后所依赖的

javenfeng, 2015-12-09:
@aa65535 请教两个问题：
1.在使用ps4玩游戏时，由于很多游戏都是udp的，所以要开启udp中继模式。那么问题来了，这时nat类型为3（xbox one下nat为严格），听说新版本修正这个问题，请问是哪个版本？
2.pandorabox安装新版shadowsocks提示内核不正确，那么，安装官方15.05的openwrt可以解决吗？

wongsyrone, 2015-12-09:
> @aa65535 <https://github.com/aa65535> @wongsyrone
> <https://github.com/wongsyrone> @SpecterHi <https://github.com/SpecterHi>

内核如果只是最后的补丁版本号不同，其实带参数强制安装是没什么问题的，如果差别太大就不要强制安装。因为正式发布的版本不会做较大改动，所以能满足内核版本要求，除非是为了解决安全隐患。

CC是稳定版本，所以基于CC编译的应该没什么问题

mie1987, 2015-12-22:
@javenfeng 请问一下你装了官方15.05的openwrt了吗，我用15.05装最新的shadowsocks还内核错误啊

javenfeng, 2015-12-22:
@mie1987 是的，内核错误。我刷回潘多拉了。希望赶快出现小米mini可以使用的ss本地客户端。