shadowsocks/openwrt-shadowsocks #78: 请求用手机 4G 网络 PPTP 接入 Openwrt 后，再通过路由内的 ss 穿透的方法

jamry, 2015-05-28:
现时情况：
境外有一VPS部署了Shadowsocks，工作正常；
家里的路由装了openwrt，部署好shadowsocks+chinadns、PPTP server，家里的设备通过路由器的LAN口或wifi已实现完美穿透。
手头的iphone（无法越狱）通过4G网络，pptp能正常连入家里路由器的VPN，并且访问国内站点正常，通过网站查看访问IP已经是家里路由器的地址，但屏蔽网站、推特、you2客户端却无法访问。

手机pptp连去VPS基本没戏，干扰太严重了，一会儿就断；连国内的pptp倒是还算稳定。然而Shadowsocks app客户端使用起来不算便利，而且不能稳定支持twitter、you2的APP。

现时路由器的firewall.user是这样设的（ppp1是pptp接入客户端）：
iptables -A input_rule -i ppp1 -j ACCEPT
iptables -A output_rule -o ppp1 -j ACCEPT

iptables -A forwarding_rule -i ppp1 -j ACCEPT
iptables -A forwarding_rule -o ppp1 -j ACCEPT

iptables -t nat -I POSTROUTING -o ppp1 -j MASQUERADE

shadowsocks用的是shadowsocks-libev-spec for openwrt优化版

请问应该怎么设置才能实现手机4G网透过家里路由的ss中转实现穿透？ thanks

aa65535, 2015-05-29:
因为你的 PPTP server 流量并没有通过 shadowsocks.
可以参考一下：
https://www.kazamigk.com/archives/241.html

closed 2015-05-29 by aa65535

cjh-github, 2015-06-08:
有没有办法让 PPTP server 流量也能区分国内国外流量？

https://www.kazamigk.com/archives/241.html 这里只说TCP流量全部重定向

iptable -t nat -I PREROUTING -p tcp -i ppp+ -j REDIRECT –to-ports 8118

cjh-github, 2015-06-08:
晚上折腾了一下，搞定了，
在Chain prerouting_rule    /* user chain for prerouting */ 中添加2条 匹配PPTP流量

iptables -t nat -A prerouting_rule -i pppoe-wan -p tcp -j RETURN
iptables -t nat -A prerouting_rule -i ppp+  -p tcp --dport 0:1024 -j SS_SPEC_LAN_AC

然后手机端PPTP连入后可以根据chnroute来透明代理

添加后重启 /etc/init.d/shadowsocks restart 会有报错 iptables: Too many links.
需要在重启前清空prerouting_rule ： iptables -t nat -F prerouting_rule

665465, 2015-09-04:
我在ubuntu上建立的pptp的server以及ss-redir的透明代理连接到国外的VPS（ipv6）,我在nat的output链将所有流量发到1080端口本机可以上网。
建立pptp server之后在iptables的加了这条，手机连pptp就能上网了（192.168.0.0/24 是pptp的网段）
iptables -t nat -A POSTROUTING -s 192.168.0.0/24 -o eth0 -j MASQUERADE
当然手机还是翻不了墙，流量没走ss
然后又加了这个
iptables -t nat -A PREROUTING -p tcp -i ppp0 -j REDIRECT --to-ports 1080
手机就上不了网了，什么网页都打不开
换成8118 port（我用privoxy转成http的代理），也是不行
iptables -t nat -A PREROUTING -p tcp -i ppp0 -j REDIRECT --to-ports 8118
我对这块不熟，能不能提供点解决思路？

nanxiaobei, 2016-03-21:
@cjh-github 你好，请问这两条规则是添加在 /etc/firewall.user 里吗？

cjh-github, 2016-03-21:
加到哪其实都无所谓，加的进就行，
新版的shadowsocks 匹配流量转发的表链名字已经改成SS_SPEC_WAN_AC，
要根据实际情况修改。

添加后可以用 iptables -L  prerouting_rule -nv 看数据包有没有匹配到，
更详细信息可以先匹配 -J LOG 后查看

nanxiaobei, 2016-03-22:
@cjh-github 
在 putty 里输入 iptables -L prerouting_rule -nv 测试显示 No chain/target/match by that name.
不过用手机连上 pptp 测试，貌似已经分流成功了，国内显示电信 ip，并且可以打开外网，多谢~！

665465, 2016-03-22:
要确认ss-redir的监听地址是0.0.0.0，否则别的接口过来的流量走不了代理。。。。

nanxiaobei, 2016-03-22:
根据我的测试，

iptables -t nat -A prerouting_rule -i pppoe-wan -p tcp -j RETURN
iptables -t nat -A prerouting_rule -i ppp+ -p tcp --dport 0:1024 -j SS_SPEC_WAN_AC

这两条规则添加在 /etc/firewall.user 里并没有效果，
而是要在 putty 中执行，才有效果。

但是这就带来一个问题，每次重启路由器后，就得重新执行一次，
所以正确方法应该是添加在 /etc/rc.local 里，重启后路由器会自动执行一遍。

另外发现，使用 luci-app-pptpd，在设置界面取消勾选「启用 MPPE 加密」然后保存&应用，手机连接 vpn 就不能打开外网了（国内网站还能打开），只能重启路由器，才恢复正常。

目前还不知道是怎么回事。

timmyludi, 2016-03-31:
你好 请问我怎么得知shadowsocks 匹配流量转发的表链名字 因为我用命令iptables -t nat -A prerouting_rule -i ppp+ -p tcp --dport 0:1024 -j SS_SPEC_WAN_AC 是错误的 应该是ss版本不一样

nanxiaobei, 2016-03-31:
@timmyludi 不是 SS_SPEC_WAN_AC 就是 SS_SPEC_LAN_AC 吧，其它的我也不清楚了。

我的 ss 版本是 2.4.5

cjh-github, 2016-04-01:
iptables -t nat -L PREROUTING
iptables -t nat -L OUTPUT

都可查看引用的表链名字
或者直接在LUCI web界面，状态》防火墙查看

timmyludi, 2016-04-01:
感谢您的回复 对这块实在是不太懂 路由器刷的明月的固件 集成了ss 以下是我运行命令的结果 请帮我看下
root@OpenWrt:~# iptables -t nat -L PREROUTING
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination
delegate_prerouting  all  --  anywhere             anywhere
REDIRECT   tcp  --  anywhere             anywhere             match-set gfwlist                                                                                                                                                              dst redir ports 10800

root@OpenWrt:~# iptables -t nat -L OUTPUT
Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination

cjh-github, 2016-04-01:
@timmyludi 你这个没有单独写表链，直接就转代理端口了，
REDIRECT tcp -- anywhere anywhere match-set gfwlist dst redir ports 10800
模仿着写一条咯，其实我也不太会写，你 iptables -t nat -S PREROUTING 导出来看就能抄着写

timmyludi, 2016-04-01:
谢谢 这回倒是查到了 shadowsocks_pre 应该是这个 然后我就 

iptables -t nat -A prerouting_rule -i ppp+ -p tcp --dport 0:1024
 -j shadowsocks_pre
这回倒是没有错误产生 使用 iptables -L prerouting_rule -nv 测试依然显示 No chain/target/match by that name 这个和上面的兄弟一样 现在还是wifi连路由的可以翻墙 pptp连路由的只能打开国内网页 小白崩溃了 哈哈

cjh-github, 2016-04-02:
@timmyludi 有时间开个SSH 我给你试试？ cjh328#gmail.com

retanoj, 2016-05-06:
我做了个实验，centos 6 client pptp连接到pptp server，server做了redir和iptables。
结论是，client无法自动配置默认路由，只有修改默认路由，才能正常翻墙

Jeffrey95, 2016-06-21:
'0:1024' 里这个1024端口代表的是什么呢?
我已经配置好了pptp和ss，通过wifi或直连到路由都可以实现自动翻墙。
可是远程通过pptp登录到vpn上，也是只能上国内的网站。即使是电脑上开启了ss客户端也翻不了。
我尝试了上面说的：
```
iptables -t nat -A prerouting_rule -i pppoe-wan -p tcp -j RETURN
iptables -t nat -A prerouting_rule -i ppp+ -p tcp --dport 0:1024 -j SS_SPEC_WAN_AC
```
也还是不行。
这是我防火墙的自定义规则
```
iptables -I INPUT -p tcp --dport 1723 -j ACCEPT
iptables -I forwarding_rule -s 192.168.1.0/24 -j ACCEPT
iptables -I INPUT -p gre -j ACCEPT
iptables -t nat -A POSTROUTING -s 192.168.1.0/24 -j MASQUERADE
```
不知道是不是我的配置有关，我防火墙里监的是1723和47(gre)端口。

koakon, 2016-09-13:
iptables -t nat -A  prerouting_wan_rule  -p tcp  -j SS_SPEC_WAN_AC
添加这条规则就可以用。
防火墙设置里面的wan-reject  input要设置为accept？或者是增加规则，pptp-lan accept。
不知道这对你们有没有用。
我这二级路由，在wan口搞ss代理。