shadowsocks/openwrt-shadowsocks #81: ss-tunnel在openwrt上无法自启动

lulooper, 2015-06-06:
shadowsocks-libev版里的ss-tunnel已经添加进自启动，但是无法自启动，ps看不到进程，但是有/var/run下有ss-tunnel.pid。手动运行是可以成功的，在rc.local里加入也启动不了，但是多加一句sleep 5就可以启动了。
root@OpenWrt:~# cat /etc/init.d/shadowsocks 
#!/bin/sh /etc/rc.common

START=95

SERVICE_USE_PID=1
SERVICE_WRITE_PID=1
SERVICE_DAEMONIZE=1

CONFIG=/etc/shadowsocks.json

start() {
        #service_start /usr/bin/ss-local -c $CONFIG -b 0.0.0.0
        #service_start /usr/bin/ss-redir -c $CONFIG -b 0.0.0.0
        service_start /usr/bin/ss-tunnel -c $CONFIG -b 0.0.0.0 -l 5353 -L 8.8.8.8:53 -u
}

stop() {
        #service_stop /usr/bin/ss-local
        #service_stop /usr/bin/ss-redir
        service_stop /usr/bin/ss-tunnel
}
root@OpenWrt:~#

closed 2015-06-08 by aa65535

griffinqiu, 2016-04-18:
需要加sleep 5吗？ 我在OpenWRT 运行Stop也不能关掉ss-redir

griffinqiu, 2016-04-18:
@lulooper @aa65535

cokebar, 2016-04-18:
try a different local port

griffinqiu, 2016-04-18:
试了几个local port，一样的。

``` bash
root@Griffins-WRT:~# /etc/init.d/shadowsocks-libev start
root@Griffins-WRT:~# ps | grep shadowsocks
  912 root      2472 S    /usr/bin/ss-redir -c /etc/shadowsocks.json -b 0.0.0.0 -f /var/run/shadowsocks.pid
  913 root      2488 S    /usr/bin/ss-tunnel -c /etc/shadowsocks.json -b 0.0.0.0 -u -l 5300 -L 208.67.220.220:53
  919 root      1304 S    grep shadowsocks
root@Griffins-WRT:~# /etc/init.d/shadowsocks-libev stop
root@Griffins-WRT:~# ps | grep shadowsocks
  912 root      2472 S    /usr/bin/ss-redir -c /etc/shadowsocks.json -b 0.0.0.0 -f /var/run/shadowsocks.pid
  925 root      1304 S    grep shadowsocks
```

lulooper, 2016-04-18:
这是有点久的一个issue了，记不太清楚了，映象中是需要sleep 5的，主要原因是在rc.local中运行有些依赖还没有起来，或者自己在sheel里写个初始化函数也行。

aa65535, 2016-04-18:
`/etc/init.d/shadowsocks-libev` 里面是什么内容？

griffinqiu, 2016-04-19:
··· bash
root@Griffins-WRT:~# cat /etc/init.d/shadowsocks-libev
#!/bin/sh /etc/rc.common

START=95

SERVICE_USE_PID=1
SERVICE_WRITE_PID=1
SERVICE_DAEMONIZE=1
SERVICE_PID_FILE=/var/run/shadowsocks.pid
CONFIG=/etc/shadowsocks.json
DNS=208.67.220.220:53
TUNNEL_PORT=5300

start() {
        # Client Mode
        # service_start /usr/bin/ss-local -c $CONFIG -b 0.0.0.0 -f $SERVICE_PID_FILE
        # Proxy Mode
        service_start /usr/bin/ss-redir -c $CONFIG -b 0.0.0.0 -f $SERVICE_PID_FILE
        # Tunnel
        service_start /usr/bin/ss-tunnel -c $CONFIG -b 0.0.0.0 -u -l $TUNNEL_PORT -L $DNS
}

stop() {
        # Client Mode
        # service_stop /usr/bin/ss-local
        # Proxy Mode
        service_stop /usr/bin/ss-redir
        # Tunnel
        service_stop /usr/bin/ss-tunnel
```

aa65535, 2016-04-19:
@griffinqiu 把参数中的 ` -f $SERVICE_PID_FILE` 去掉

griffinqiu, 2016-04-19:
一样的 ， 奇怪。算了不整了，这个也没啥子影响
``` bash
root@Griffins-WRT:~#  /etc/init.d/shadowsocks-libev start
root@Griffins-WRT:~# ps | grep shad
 8705 root      2472 S    /usr/bin/ss-redir -c /etc/shadowsocks.json -b 0.0.0.0
 8708 root      2488 S    /usr/bin/ss-tunnel -c /etc/shadowsocks.json -b 0.0.0.0 -u -l 5300 -L 208.67.220.220:53
 8710 root      1304 S    grep shad
root@Griffins-WRT:~#  /etc/init.d/shadowsocks-libev stop
root@Griffins-WRT:~# ps | grep shad
 8705 root      2472 S    /usr/bin/ss-redir -c /etc/shadowsocks.json -b 0.0.0.0
 8716 root      1304 S    grep shad
```

aa65535, 2016-04-19:
service_stop 不行的话就用 `killall -q -9 ss-redir` 吧。

arfaWong, 2016-04-19:
同样遇到这个问题，只能通过 sleep 5 来解决了。

heiguo, 2016-06-16:
同样遇到这个问题，在 OpenWRT 中，log 如下：
`Wed Jun 15 15:52:29 2016 kern.err /usr/bin/ss-tunnel[1066]: [udp] getaddrinfo: ai_family not supported
Wed Jun 15 15:52:29 2016 kern.err /usr/bin/ss-tunnel[1066]: [udp] bind() error`