shadowsocks/openwrt-shadowsocks #82: ss-redir已经支持udp了，还需要ss-tunnel吗？

suikatomoki, 2015-06-07:
有没有精简掉ss-tunnel的计划，毕竟功能重复了。还说是我理解有偏差，ss-tunnel是非要不可的？

xcjiang, 2015-06-08:
巧了，我刚刚也在想这个问题，我感觉应该可以涵盖ss-tunnel.

aa65535, 2015-06-08:
ss-tunnel 是提供给本地DNS查询使用的。

closed 2015-06-08 by aa65535

xcjiang, 2015-06-08:
现在ss-tunnel的用途就是把本地dns的请求转发到一个墙外的dns，比如8.8.8.8，那么如果ss-redir可以直接透明转发dns的udp包，那么本地dns可以直接使用8.8.8.8，而无需建一个tunnel，我是这样理解的，不知道对不对。

aa65535, 2015-06-09:
ss-redir 的 TCP 和 UDP 流量在 iptables 中的处理不在同一个表里，因为使用的是 TPROXY ，无法代理自身的。

suikatomoki, 2015-06-09:
明白了

perol, 2016-04-23:
为什么使用TPROXY，UDP可以跟TCP一样使用 REDIRECT 吗？

Mattosx, 2016-07-13:
@aa65535 我目前安装 模块后，开启-u 是可以自动转发dns的，配合chinadns使用，和ss-tunnel功能一样。

wendellou, 2016-08-05:
tunnel 和 redirect语意以及实现的效果不同吧

gonejack, 2016-08-22:
ss-redir只是本地转发UDP，无法做到代理，所以只能用来转发QQ消息或者国内DNS查询之类的墙内UDP。

真相就是这样咯，想DNS查询8.8.4.4之类被墙的IP，还是得用ss-tunnel。

rav009, 2016-12-09:
ss-tunnel 是不是就是luci里的 UDP Forewarding？

smthvitas, 2017-02-01:
@aa65535 "ss-tunnel 是提供给本地DNS查询使用的。"  意思是说ss-redir可以用于局域网，但不能用于本机，所以ss-tunnel还是有必要的是吗？
这么说"ss-redir只是本地转发UDP，无法做到代理"有误，路由器上用ss-redir也就够了是吧。

smthvitas, 2017-02-01:
此外，udp转发必须用TPROXY的本质原因是不是因为，UDP没有类似TCP的ip_conntrack()函数能够获取到REDIRECT之前的目标IP和端口，所以必须把原ip报文原封不动转发到本地特定socket？