shadowsocks/openwrt-shadowsocks #88: 在非openwrt环境下使用iptables转发

jingzhaoyang, 2015-06-26:
我这里使用一台linux机器做网关，我参考了shadwosocks-libev的文档，配置了ss-redir的透明代理转发不成功，不知道问题出在什么地发。您能帮我查找下原因吗？或者有什么思路去处理？
下面是我的环境：
ech0:192.168.1.40(模拟wan口)
eth1:172.16.16.1(lan口)
iptables 规则表

[root@ha-master ~]# iptables -L -t nat -n
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         
SHADOWSOCKS  tcp  --  0.0.0.0/0            0.0.0.0/0           

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
MASQUERADE  all  --  172.16.16.0/24       0.0.0.0/0           

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain SHADOWSOCKS (1 references)
target     prot opt source               destination      
#这里xxx.xxx.xxx.xxx是我服务器端地址   
RETURN     all  --  0.0.0.0/0            xxx.xxx.xxx.xxx 
RETURN     all  --  0.0.0.0/0            0.0.0.0/8           
RETURN     all  --  0.0.0.0/0            10.0.0.0/8          
RETURN     all  --  0.0.0.0/0            127.0.0.0/8         
RETURN     all  --  0.0.0.0/0            169.254.0.0/16      
RETURN     all  --  0.0.0.0/0            172.16.0.0/12       
RETURN     all  --  0.0.0.0/0            192.168.0.0/16      
RETURN     all  --  0.0.0.0/0            224.0.0.0/4         
RETURN     all  --  0.0.0.0/0            240.0.0.0/4         
REDIRECT   tcp  --  0.0.0.0/0            0.0.0.0/0           redir ports 12345 

此时，本机没有使用代理，而172.16.16.10内网的机器也无法上网。

当我把SHADOWSOCKS应用导OUTPUT链路时，我本机可以走代理上网，但是nat过来的机器能上网但是不走代理。
iptables -t nat -A PREROUTING -p tcp -j SHADOWSOCKS

aa65535, 2015-06-27:
使用 `iptables-save` 打印你的 iptables 规则。
要正常使用，需要先将这个Linux机器配置成可以正常工作的网关。

jingzhaoyang, 2015-06-29:
```
# Generated by iptables-save v1.4.7 on Fri Jun 26 12:56:03 2015
*mangle
:PREROUTING ACCEPT [6307:678622]
:INPUT ACCEPT [4817:492427]
:FORWARD ACCEPT [660:91778]
:OUTPUT ACCEPT [3317:401208]
:POSTROUTING ACCEPT [3977:492986]
:SHADOWSOCKS - [0:0]
-A PREROUTING -j SHADOWSOCKS
-A SHADOWSOCKS -p udp -m udp --dport 53 -j TPROXY --on-port 12345 --on-ip 0.0.0.0 --tproxy-mark 0x0/0x0
COMMIT
# Completed on Fri Jun 26 12:56:03 2015
# Generated by iptables-save v1.4.7 on Fri Jun 26 12:56:03 2015
*nat
:PREROUTING ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:SHADOWSOCKS - [0:0]
-A PREROUTING -p tcp -j SHADOWSOCKS
-A POSTROUTING -s 172.16.16.0/24 -o eth0 -j MASQUERADE
-A SHADOWSOCKS -d 128.199.xxx.xxx/32 -j RETURN
-A SHADOWSOCKS -d 0.0.0.0/8 -j RETURN
-A SHADOWSOCKS -d 10.0.0.0/8 -j RETURN
-A SHADOWSOCKS -d 127.0.0.0/8 -j RETURN
-A SHADOWSOCKS -d 169.254.0.0/16 -j RETURN
-A SHADOWSOCKS -d 172.16.0.0/12 -j RETURN
-A SHADOWSOCKS -d 192.168.0.0/16 -j RETURN
-A SHADOWSOCKS -d 224.0.0.0/4 -j RETURN
-A SHADOWSOCKS -d 240.0.0.0/4 -j RETURN
-A SHADOWSOCKS -p tcp -j REDIRECT --to-ports 12345
COMMIT
# Completed on Fri Jun 26 12:56:03 2015
# Generated by iptables-save v1.4.7 on Fri Jun 26 12:56:03 2015
*filter
:INPUT ACCEPT [17106:14659251]
:FORWARD ACCEPT [668:92450]
:OUTPUT ACCEPT [13475:3979910]
COMMIT
# Completed on Fri Jun 26 12:56:03 2015
```
这是我的iptables-save 保存的文件，其中128.199.xxx.xxx是我墙外的机器。

aa65535, 2015-06-29:
@jingzhaoyang 看规则是没问题的，如果不开 shadowsocks 时是可以正常联网的话，你需要看看 shadowsocks 配置的本地端口(local_port)是不是 12345.

jingzhaoyang, 2015-07-03:
@aa65535 谢谢你的帮助，最近忙一直没顾的上回复。端口配置都是正确的，但是就是实现不了。现在放弃通过iptables转发端口了，换了shadowvpn方案。

closed 2015-07-03 by jingzhaoyang