shadowsocks/openwrt-shadowsocks #90: 关于ipset导入IP段效率(非OpenWRT平台)

houzi-, 2015-06-29:
您好！
我写了个成批量导入IP段的脚本，感觉导入时间有点长，跟op下的没法比，特此来请教一下该如何优化！
```shellscript
#!/bin/sh
# Copyright (C) 2015 OpenWrt.org

_input=/jffs/chnroute.txt
IPS=/usr/sbin/ipset

ipset -N gfwlist nethash

egrep -v "^#|^$" $_input | while IFS= read -r ip

do
        $IPS -A gfwlist $ip 
done
```

aa65535, 2015-06-30:
通过循环逐条添加，效率上肯定会慢很多，可以使用 `ipset -R` 批量添加。

```sh
#!/bin/sh
# Copyright (C) 2015 OpenWrt.org

_input=/jffs/chnroute.txt
IPS=/usr/sbin/ipset

$IPS -! -R <<EOF
create gfwlist hash:net
$(egrep -v "^#|^$" $_input | sed -e "s/^/add gfwlist /")
EOF
```

closed 2015-06-30 by aa65535

houzi-, 2015-07-02:
@aa65535 
执行后出现:
```shellscript
root@unknown:/tmp/home/root# /jffs/ip.sh
ipset v4.5: Unknown arg `-!'
Try `ipset -H' or 'ipset --help' for more information.
```
ipset 4.5没有这个 ' ! '参数！

aa65535, 2015-07-02:
试试将 ipset -! -R 换成 ipset -exist restore ，不行的话就去掉-exist选项吧。

https://github.com/shadowsocks/openwrt-shadowsocks/issues/90#issuecomment-117999472

houzi-, 2015-07-02:
@aa65535 
还是不行！两个都测试了
我是这样改的:
$IPS --restore <<EOF
出现以下错误:
```shell
ipset v4.5: Line 1 does not start as a valid restore command

Restore failed at line 1:
Try `ipset -H' or 'ipset --help' for more information.
```

aa65535, 2015-07-02:
restore 前面没有 -- 的。

https://github.com/shadowsocks/openwrt-shadowsocks/issues/90#issuecomment-118005918

houzi-, 2015-07-02:
@aa65535 
加了还是执行不了！
出现:
ipset v4.5: Unknown arg `ipset'

aa65535, 2015-07-02:
@houzi- 你不会写成 `ipset ipset restore` 了吧？

houzi-, 2015-07-03:
@aa65535 
ipset v4.5

Usage: ipset -N new-set settype [options]
       ipset -[XFLSH] [set] [options]
       ipset -[EW] from-set to-set
       ipset -[ADT] set IP
       ipset -R
       ipset -v
       ipset -h (print this help information)

Commands:
Either long or short options are allowed.
  --create  -N setname settype <options>
                    Create a new set
  --destroy -X [setname]
                    Destroy a set or all sets
  --flush   -F [setname]
                    Flush a set or all sets
  --rename  -E from-set to-set
                    Rename from-set to to-set
  --swap    -W from-set to-set
                    Swap the content of two existing sets
  --list    -L [setname] [options]
                    List the IPs in a set or all sets
  --save    -S [setname]
                    Save the set or all sets to stdout
  --restore -R [option]
                    Restores a saved state
  --add     -A setname IP
                    Add an IP to a set
  --del     -D setname IP
                    Deletes an IP from a set
  --test    -T setname IP
                    Tests if an IP exists in a set.
  --help    -H [settype]
                    Prints this help, and settype specific help
  --version -V
                    Prints version information

Options:
  --sorted     -s   Numeric sort of the IPs in -L
  --numeric    -n   Numeric output of addresses in a -L (default)
  --resolve    -r   Try to resolve addresses in a -L
  --quiet      -q   Suppress any output to stdout and stderr.

没有重复写！好诡异！

aa65535, 2015-07-03:
@houzi- 这样应该是可以的

```sh
#!/bin/sh
# Copyright (C) 2015 OpenWrt.org

_input=/jffs/chnroute.txt
IPS=/usr/sbin/ipset

$IPS -R <<EOF
create gfwlist hash:net
$(egrep -v "^#|^$" $_input | sed -e "s/^/add gfwlist /")
EOF
```

houzi-, 2015-07-03:
@aa65535 
上次我就这样改，也是报错！
ipset v4.5: Line 1 does not start as a valid restore command

Restore failed at line 1:
Try `ipset -H' or 'ipset --help' for more information.

aa65535, 2015-07-03:
@houzi-
ipset 从 v5.0 的时候重写过，可能一些语法不一样。
你先手动添加几条至 gfwlist 这个set 中，然后使用 `ipset -S gfwlist` 打印出它保存的规则，然后按照那个规则修改即可。

houzi-, 2015-07-03:
# Generated by ipset 4.5 on Thu Jan  1 08:05:04 1970
-N gfwlist nethash --hashsize 39366 --probes 4 --resize 50
-A gfwlist 103.237.244.0/22
-A gfwlist 1.2.6.0/23
-A gfwlist 43.250.168.0/22
COMMIT
# Completed on Thu Jan  1 08:05:04 1970
应该没错！

aa65535, 2015-07-03:
@houzi- 原来是 iptables 风格的，试试

```sh
#!/bin/sh
# Copyright (C) 2015 OpenWrt.org

_input=/jffs/chnroute.txt
IPS=/usr/sbin/ipset

$IPS -R <<EOF
-N gfwlist nethash
$(egrep -v "^#|^$" $_input | sed -e "s/^/-A gfwlist /")
COMMIT
EOF
```

houzi-, 2015-07-03:
OK! 
原来也是这样改了-N gfwlist nethash这行，但是没加应用保存命令！
谢谢啊！

muronghan, 2016-11-27:
@aa65535 特别向你请教这个可执行文件的写法。我用的是ipset v6.24，在OpenWrt里，想通过命令自动添加所有中国的IP地址段到ipset里，而不是一条一条写。

目前只在IP地址上做成功了，所以我的问题是如果是IP地址段，该怎么写？以下是我的执行文件，之前已经先在/etc/firewall.user里建立了chinaip这个ipset （ipset create chinaip hash:ip)

    _input=/etc/ignore.list
    IPS=/usr/sbin/ipset
    egrep -v "^#|^$" $_input | while IFS= read -r ip
    do
        $IPS -A chinaip $ip
    done

aa65535, 2016-11-27:
@muronghan 

```sh
_input=/etc/ignore.list
IPS=/usr/sbin/ipset
$IPS -! restore <<-EOF || return 1
    create chinaip hash:ip
    $(egrep -v "^#|^$" $_input)
EOF
```

muronghan, 2016-11-27:
很感谢，可能我没有说清楚，我要加入ipset的是IP地址段，即像1.0.1.0/24这种格式，所以用您的命令，结果输出是：

ipset v6.24: No command specified: unknown argument 1.0.1.0/24

aa65535, 2016-11-27:
@muronghan 

```sh
_input=/etc/ignore.list
IPS=/usr/sbin/ipset
$IPS -! restore <<-EOF || return 1
	create chinaip hash:net
	$(egrep -v "^#|^$" $_input | sed -e "s/^/add chinaip /")
EOF
```

muronghan, 2016-11-27:
@aa65535 这次呈现：

ipset v6.24: No command specified: unknown argument EOF