shadowsocks/openwrt-shadowsocks #91: 关于允许WAN IP走代理或者忽略问题

houzi-, 2015-06-30:
您好！
我该如何添加条规则让WAN IP走代理或者忽略而不冲突！
脚本:
```shellscript
#!/bin/sh
# Copyright (C) 2007-2012 OpenWrt.org

iptables -t nat -N SHADOWSOCKS
iptables -t nat -A SHADOWSOCKS -d 0.0.0.0/8 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 10.0.0.0/8 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 127.0.0.0/8 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 169.254.0.0/16 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 172.16.0.0/12 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 192.168.0.0/16 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 224.0.0.0/4 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 240.0.0.0/4 -j RETURN
iptables -t nat -A SHADOWSOCKS -d 157.7.109.114 -j RETURN
iptables -t nat -A SHADOWSOCKS -p tcp -m set --set gfwlist dst -j RETURN
iptables -t nat -A SHADOWSOCKS -p tcp -j REDIRECT --to-ports 1080
iptables -t nat -I PREROUTING -p tcp -m multiport --dports 22,80,443 -j SHADOWSOCKS
```
望您能够解答一下！

aa65535, 2015-06-30:
如果你想实现这个功能的话可以参考 [ss-rules](https://github.com/shadowsocks/openwrt-shadowsocks/blob/master/files/shadowsocks.rule) .即使在非OpenWRT平台，在不使用 `-c` 参数时它也是可以工作的。

BTW: 这里不是用来提问题的地方，有问题可以到一些论坛寻求帮助。

closed 2015-06-30 by aa65535

houzi-, 2015-07-13:
@aa65535 
把ss-rules应用到TT上出现:
```shell
/opt/bin/ss-rules: line 165: getopts: not found
···
好像是不支持getopts命令！