shadowsocks/openwrt-shadowsocks #94: openwrt shadowsocks UDP代理不能启动

linuxchang, 2015-07-01:
提示
[image]

但是如果在前段页面上面修改为  TCP-only   就没有问题，但是我的环境是希望能够代理UDP，希望您能帮忙解答一下
[image]

我的安装包是
[image]

Luci的版本是
[image]

我现在的情况是必须要选择TCP-only,要是选择TCP+UDP 每次重启路由器shadowsocks就起不来

linuxchang, 2015-07-01:
而且我在路由器里面看了，Iptable里面只有关于TCP的条目，没有关于UDP的

linuxchang, 2015-07-01:
现在的情况是，如果我在LUCI界面上面选择  TCP only  那么 shadowsocks  启动是没有问题的

linuxchang, 2015-07-02:
通过观察启动的脚本发现是rules 脚本里面的这句规则
[image]

aa65535, 2015-07-02:
有安装 `iptables-mod-tproxy` ？

linuxchang, 2015-07-02:
我安装了。。 
我用的固件是
[image]
其余的包
[image]
[image]
通过产看https://github.com/shadowsocks/openwrt-shadowsocks/blob/master/files/shadowsocks.rule的规则
发现就是TPROXY的规则除了问题
[image]
我在路由器里面也尝试了一下  如果指定 -A   无论是 OUTPUT 还是POSTROUTING  都是一样的错误

aa65535, 2015-07-02:
安装后重启路由器了吗？

https://github.com/shadowsocks/openwrt-shadowsocks/issues/94#issuecomment-117951351

linuxchang, 2015-07-02:
我重启了很多次了！  我现在怀疑是iptables 的这个模块有问题，希望您能帮助我一下

linuxchang, 2015-07-02:
另外我还有一个问题想问您，我的SS后台用的是这个
https://github.com/mengskysama/shadowsocks/tree/manyuser
用这个是因为他多用户，这个版本说也能支持UDP，但是我不清楚，他说的这个UDP代理很您开发的这个版本有什么区别，网上说只有您开发的libev  才支持UDP代理

aa65535, 2015-07-02:
跟服务端没关系的，使用 lsmod 命令看看内核模块有没有加载成功。

linuxchang, 2015-07-02:
[image]

然后我把关于TPROXY的2个包重装了，然后重启了  再次执行测试

[image]

madeye, 2015-07-02:
`TPROXY` only works on `PREROUTING` chain.

linuxchang, 2015-07-02:
现在看来跟指定谁好像没有关系。。。。。

[image]

linuxchang, 2015-07-02:
就是报错，哎。。。

[image]

aa65535, 2015-07-02:
内核模块名称应该是 xt_TPROXY.
使用 iptables-save -t mangle 看看这个表里有什么规则。

linuxchang, 2015-07-02:
现在来看应该是模块还是没有。。
[image]

mangle规则
[image]

启动shadowsocks后的规则
[image]

aa65535, 2015-07-02:
内核模块没加载成功是没法使用这个功能的。
看看有没有漏掉什么包，内核日志里面看看有没有错误。

linuxchang, 2015-07-02:
好的！ 因为这个版本的固件自带了很多我不需求的东西，我就都给删了，可能是这个原因，我打算重新刷了。非常感谢您的耐心指导。谢谢

closed 2015-07-04 by aa65535