shadowsocks/openwrt-shadowsocks #95: chaos_calmer rc2 无法使用

1ternal, 2015-07-01:
我使用的是 [qihoo c301](http://wiki.openwrt.org/toh/qihoo/c301) 这款路由器。
- 安装 spec 版报错。信息如下：
```shell
Installing shadowsocks-libev-spec (2.2.2-1) to root...
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipt-conntrack and kmod-ipt-conntrack) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipv6 and kmod-ipv6) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ip6tables and kmod-ip6tables) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for shadowsocks-libev-spec:
 * 	kernel (= 3.18.16-1-96841099aee8a14eb2ea956c15a2194e) * 	kernel (= 3.18.16-1-96841099aee8a14eb2ea956c15a2194e) * 	kernel (= 3.18.16-1-96841099aee8a14eb2ea956c15a2194e) *
 * opkg_install_cmd: Cannot install package shadowsocks-libev-spec.
```

- 安装非 spec 版，可以安装。`ps` 无进程。手动启动，报错如下：
```shel
ss-redir: can't load library 'libc.so'
```

求帮助，求解决...

aa65535, 2015-07-02:
- 依赖问题参考 #70

 - 先确认 libc 包已经安装，`ln -s /lib/libuClibc-0.9.33.2.so /lib/libc.so`

closed 2015-07-02 by aa65535

1ternal, 2015-07-02:
@aa65535 多谢回复 :smile: 
应该是 package list 的问题。
不保留配置，安装 RC2 就没问题。