shadowsocks/openwrt-shadowsocks #96: Allow using different IP and ports for TCP and UDP

clowwindy, 2015-07-06:
Maybe it's dangerous to use the same port.

@aa65535 @madeye

madeye, 2015-07-06:
Two `ss-redir` instances should solve the problem.

clowwindy, 2015-07-06:
We can add a new configuration entry in LuCI for the UDP settings like this

```
TCP:
- Server:
- Port:
- ...

UDP:
[ ] Use the same as TCP
- Server:
- Port:
- ...
```

aa65535, 2015-07-10:
This is necessary?

clowwindy, 2015-07-10:
Some people report that by disabling UDP, they got less chance that their servers get blocked.

cokebar, 2015-07-14:
UDP is much more unstable. I cannot use TCP+UDP mode on my router. I have a high chance encountering "QUIC protocol error" when using chrome browser with TCP+UDP mode. I have to wait for about 1 min, and the connection recovers. This is definitely GFW's temporary block.
I think UDP connection have a higher chance that the server get blocked.
Mabe using a different port for UDP can protect server's TCP port.

commit e8eb9013 (2015-07-22): bump version to 2.2.3

close #96
closed 2015-07-22 by aa65535