shadowsocks/openwrt-shadowsocks #98: OpenWRT:CHAOS CALMER (15.05-rc2, r45918):failed to find a module named nf_tproxy_core

hackevin, 2015-07-06:
Configuring ip.
Configuring zlib.
Configuring libopenssl.
Configuring resolveip.
Configuring kmod-nfnetlink.
Configuring kmod-ipt-ipset.
Configuring libmnl.
Configuring ipset.
Configuring kmod-ipt-tproxy.
failed to find a module named nf_tproxy_core
Configuring iptables-mod-tproxy.
Configuring shadowsocks-libev-spec.

aa65535, 2015-07-10:
reboot

closed 2015-07-10 by aa65535

zlphoenix, 2015-08-25:
I also encountered this problem when install shadowsocks-libev package on my 3700 v2. After Reboot router , this package still not found.

Steltek, 2015-12-31:
Same error message with the released version of Chaos Calmer on ar71xx. Neither iptables-mod-tproxy_1.4.21-1_ar71xx.ipk nor kmod-ipt-tproxy_3.18.20-1_ar71xx.ipk include the nf_tproxy_core file/module. They only include libxt_socket.so and libxt_TPROXY.so. 

Not sure if nf_tproxy_core is actually needed. If it's not, the included /etc/modules.d/ipt-tproxy and postinst-pkg probably shouldn't reference it.

ZaXk, 2016-01-07:
root@OpenWrt:~# opkg install iptables-mod-tproxy
Installing iptables-mod-tproxy (1.4.21-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/mvebu/generic/packages/base/iptables-mod-tproxy_1.4.21-1_mvebu.ipk.
Installing kmod-ipt-tproxy (3.18.20-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/mvebu/generic/packages/base/kmod-ipt-tproxy_3.18.20-1_mvebu.ipk.
Configuring kmod-ipt-tproxy.
failed to find a module named nf_tproxy_core
xt_socket is already loaded
xt_TPROXY is already loaded
Configuring iptables-mod-tproxy.

Don't know how to fix it!

xdrwedc, 2016-04-19:
Reboot and install again

Step 1 reboot
Step 2 rm /etc/config/shadowsocks
Step 3 opkg install shadowsocks-libev-spec

no failures anymore

jackfengji, 2016-06-26:
UDP forwarding is not working without kmod-ipt-tproxy. But when I tried to install kmod-ipt-tproxy, it gives me the error
failed to find a module named nf_tproxy_core

yangxuan8282, 2016-12-16:
got same issues, fix with:

    opkg update 

    opkg install ip-full

oliver661, 2017-02-20:
Got same problem when "opkg install kmod-ipt-tproxy"
Reboot, reinstall and everything I found is not working.
And I found this ticket https://bugs.lede-project.org/index.php?do=details&task_id=212 said that 

> nf_tproxy_core was removed during 3.12 development...

So, I think re-flash openwrt by 15.05 version which is not remove nf_tproxy_core or LEDE system which has already fix this issue would work.