shadowsocks/openwrt-shadowsocks #99: 更新忽略IP列表问题

houzi-, 2015-07-06:
您好！
我用的不是spec，而是libev+防火墙策略，再配合ChinaDNS。（跟ss-rules类似)
用wget更新出现问题:
```shellsprict
root@OpenWrt:~# wget -O- 'http://ftp.apnic.net/apnic/stats/apnic/delegated-apnic
-latest' | awk -F\| '/CN\|ipv4/ { printf("%s/%d\n", $4, 32-log($5)/log(2)) }' >
/etc/ignore.list
--2015-07-06 21:47:27--  http://ftp.apnic.net/apnic/stats/apnic/delegated-apnic-latest
Resolving ftp.apnic.net... failed: Name or service not known.
wget: unable to resolve host address 'ftp.apnic.net'
```
自己的防火墙策咯:
```shellsprict
#!/bin/sh
# Copyright (C) 2015 OpenWrt.org

iptables -t nat -N SHADOWSOCKS
iptables -t mangle -N SHADOWSOCKS
iptables -t nat -A SHADOWSOCKS -d 0.0.0.0/8 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 10.0.0.0/8 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 127.0.0.0/8 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 169.254.0.0/16 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 172.16.0.0/12 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 192.168.0.0/16 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 224.0.0.0/4 -j RETURN 
iptables -t nat -A SHADOWSOCKS -d 240.0.0.0/4 -j RETURN
iptables -t nat -A SHADOWSOCKS -d 157.7.109.114 -j RETURN
iptables -t nat -A SHADOWSOCKS -p tcp -m set --match-set gfwlist dst -j RETURN
iptables -t nat -A SHADOWSOCKS -p tcp -j REDIRECT --to-ports 1080
ip rule add fwmark 0x01/0x01 table 100
ip route add local 0.0.0.0/0 dev lo table 100
iptables -t mangle -A SHADOWSOCKS -p udp --dport 53 -j TPROXY --on-port 1080 --tproxy-mark 0x01/0x01
iptables -t nat -A PREROUTING -p tcp -m multiport --dports 22,80,443 -j SHADOWSOCKS
iptables -t mangle -A PREROUTING -j SHADOWSOCKS
iptables -t nat -A OUTPUT -p tcp -j SHADOWSOCKS
```

suikatomoki, 2015-07-07:
ftp.apnic.net换成202.12.29.205试试

houzi-, 2015-07-07:
@suikatomoki 你想表达什么意思！

suikatomoki, 2015-07-07:
wget -O- 'http://202.12.29.205/apnic/stats/apnic/delegated-apnic
-latest' | awk -F\| '/CN\|ipv4/ { printf("%s/%d\n", $4, 32-log($5)/log(2)) }' >
/etc/ignore.list

suikatomoki, 2015-07-07:
wget -O- 'http://202.12.29.205/apnic/stats/apnic/delegated-apnic-latest' | awk -F| '/CN|ipv4/ { printf("%s/%d\n", $4, 32-log($5)/log(2)) }' >/etc/ignore.list

aa65535, 2015-07-10:
DNS设置问题
`echo "nameserver 114.114.114.114" >>/etc/resolv.conf`

closed 2015-07-10 by aa65535