shadowsocks/openwrt-shadowsocks #101: 全局代理时dns问题

dixyes, 2015-07-10:
可能是我的网络比较奇葩 ss走教育网v6到vps（tunnelbroker） 其余走v4网通 开启全局代理+udp启用ss后 所有dns查询严重变慢甚至超时导致不能上网

WNDR3800 ar7161
BARRIER BREAKER (14.07, r42625)
shadowsocks-libev-spec-polarssl - 2.2.2-1 sourceforge上的预编译版

dixyes, 2015-07-11:
而且 还有这样的bug
复现姿势：
使用一个ipv6ss服务器
首先忽略列表，tcp+udp链接
正常使用 仅仅是dns龟速
然后全局，tcp+udp
喜闻乐见的事就发生了
v4所有链接不可用
v6正常
解决姿势是 把udp转发关掉
（以上人懒都是luci里动的

Crystommy, 2015-07-12:
我也出现了这种问题，此时netstat -nlp会看到很多端口

cokebar, 2015-07-14:
首先你要测试一下教育网对于外部UDP端口的封杀情况。我们学校，国外IP的UDP几乎完全无法访问，如果你使用ss的ss-tunnel来做国外DNS解析，你收不到任何response，会卡死在DNS解析。

然后就是你不把自己的详细配置贴出来完全无法分析

aa65535, 2015-07-14:
我没有 IPv6 环境去测试这个问题，而且 ss-rules 目前也没有支持 IPv6 ，所以在连接 IPv6 时可能会造成死循环，你可能需要使用 ip6tables 命令排除掉服务器的那个 IPv6 地址。

dixyes, 2015-07-14:
1.没有封杀 这条消息通过教育网v6，搬瓦工vps发送到github（电脑直接插网线
（顺便 我用的v6ip
2.详细配置：
/etc/config/shadowsocks

config shadowsocks
        option tunnel_port '5300'
        option tunnel_forward '8.8.4.4:53'
        option use_conf_file '0'
        option server_port '2333'
        option local_port '12450'
        option timeout '60'
        option password '这里是密码'
        option encrypt_method 'aes-256-cfb'
        option udp_relay '1'
        option lan_ac_mode '0'
        option tunnel_enable '0'
        option ignore_list '/etc/shadowsocks/ignore.list'
        list wan_bp_ip '这里是服务器v4ip'
        option server '这里是v6ip'
        option enable '0' 关掉了以正常上网

3.并不懂ip6tables用法 求教如何设置排除

ohyeah521, 2015-09-24:
ipv4下，运行一段时间后，dns解析就会出错。也就是ss的dns转发出错了，不能够解析域名，曾经解析过的域名因为dnsmasq有缓存，可以翻墙。求解决。

closed 2015-11-21 by aa65535

Valentine23v, 2016-09-03:
@ohyeah521 我好像出现了和你一样的情况。

dixyes, 2016-09-03:
是我的错觉还是这是我一年前开的issue
而且问题已经修正了？
（目前安心通过v6连接 双栈（v4+natv6）都没啥问题
@Valentine23v 更新你的ss版本后若问题仍存在就再开一个新issue吧