shadowsocks/openwrt-shadowsocks #106: 运行了一段时间Shadowsocks-libev for OpenWrt 发现  ss-redir 经常莫名的假死

linuxchang, 2015-07-15:
我用的是小米mini路由器，刷的是潘多拉版本的openwrt，Shadowsocks-libev for OpenWrt 版本是2.2.1,
[image]

目前发现一个问题，ss-redir  经常无故停止工作，进程还在就是不干活了，由于启用了忽略列表，访问国内的网站是没有问题的，国外的需要翻墙的就通通不行了，这个时候只能进入路由器重启shadowsocks进程，重启之后可以正常工作一段时间，但是不一定什么时候就又不行了。

这是出问题时候的截图，我访问wiki.archlinux.org，非常缓慢。。。。
[image]

我在服务器也查看了，应该不是服务端的问题，因为每次出问题，我只需要进入路由器重启shadowsocks客户端就可以了

isbaseby, 2015-07-19:
newifi mini 也有同样问题 很久很久了

scenix007, 2015-07-21:
表示我可能也遇到同样的问题了，路由器是Netgear WNDR4300， Shadowsocks-libev-spec版本是2.2.2-1。后台查进程都在，就是不干活。有时候重启shadowsocks服务也不行，需要在luci界面里修改下服务器ip，然后才好用。不知道如何debug，也不知道哪里看日志。求指点

aa65535, 2015-07-21:
试试将 timeout 改小

scenix007, 2015-07-21:
谢谢回复，我默认的timeout是300，改成60用一段时间试试。

clowwindy, 2015-07-21:
重启浏览器能解决吗？

linuxchang, 2015-07-21:
不能，出问题的时候，只有重启进程才管用

myliyifei, 2015-07-22:
建议在linux下面使用nc 命令来排错

scenix007, 2015-07-23:
昨天晚上又出现了类似的问题，重启路由器甚至都无法解决。
看了#96这个issue，发现我也是启用了UDP转发。
不过我路由器上ss-redir不干活的时候，我pc和mac端的shaodowsocks程序都能很正常的工作。
是否有可能我也被gfw临时block了？
晚上升级一下新版看看是否还有这个问题。

linuxchang, 2015-07-23:
启用UDP转发确实会有一些影响，这个我也发现了，特别是一些视频类的网站，会持续很长时间打不开，在调试模式下经常能看到quic协议的错误

scenix007, 2015-07-23:
是的 尤其在看了一会you2be之后, 极容易挂。
我已经安装了新版，并打开了UDP中继，试试看。
另外请教楼上，如何进入调试模式？

orzsun, 2015-07-25:
用BT下一会东西后也会出现这个情况,重启路由的shadowsocks才可以解决

linuxchang, 2015-07-27:
怀疑可能是端口的问题，于是在路由器的启动项里面改了一些内核参数，这些内核参数网上都有，就是做SS后台时候修改的参数，使用了1个星期发现效果甚微。。。。看来问题不在这里

madeye, 2015-07-27:
`echo 32768 > /proc/sys/fs/file-max`

scenix007, 2015-07-27:
@madeye ss-dir假死的时候

```bash
root@OpenWrt:~# cat /proc/sys/fs/file-nr
218	0	12640
```

所以感觉不是file-max的问题

madeye, 2015-07-27:
@scenix007 你的问题和 @linuxchang 明显不是一个问题：

“昨天晚上又出现了类似的问题，重启路由器甚至都无法解决。”

scenix007, 2015-07-27:
@madeye 看起来我是被gwf临时照顾了。。。

wongsyrone, 2015-07-27:
不知道你们用没用dnsmasq的ipset，用了的话可能是dnsmasq jail的问题，可以看最近的几条官方changeset，如果没有就忽略我吧。另外个人实际测试 `echo 16384 > /proc/sys/fs/file-max` 已经足够

YuningMiao, 2015-08-01:
Shadowsocks 和shadowvpn 在我搬家以后都有这个问题，网络没事，vps可以ping通，可是ss过一阵就断了

ohyeah521, 2015-09-24:
我也出现这个问题了。求解决方案。

ohyeah521, 2015-09-24:
@wongsyrone  我dnsmasq 是 2.72 版本的，哪里有最新版本的下载链接？

oglops, 2015-09-29:
我好像也遇到这个问题，我是tomato shibby v131, 我用tomato自带的工具链编译了 2.75, 拷到jffs上，然后在administration->init script里放上
```
mount --bind /jffs/dnsmasq/dnsmasq  /usr/sbin/dnsmasq
killall dnsmasq
```
虽然能换成高版本的dnsmasq,但是好像没什么效果， 此方法是v2ex看来的

pexcn, 2015-10-07:
我也似乎遇到了这个问题，好像换了ip就好了

pexcn, 2015-11-04:
终于抓到了出问题时候的log，大神们看看

```
Thu Nov  5 00:06:56 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:06:56 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:06:56 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:06:57 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:06:57 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:06:57 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:06:57 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:06:57 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:06:57 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:06:58 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:06:58 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:06:58 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:06:58 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:06:58 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:06:58 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:06:59 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:06:59 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:06:59 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:06:59 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:00 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:00 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:01 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:01 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:01 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:01 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:02 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:02 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:02 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:02 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:02 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:02 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:02 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:02 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:02 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:02 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:03 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:03 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:03 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:03 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:04 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:04 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:05 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:05 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:06 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:06 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:06 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:06 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:06 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:06 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:07 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:07 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:07 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:07 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:07 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
[... cut ...]
Thu Nov  5 00:07:37 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:37 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:37 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
Thu Nov  5 00:07:37 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] server receive a packet
Thu Nov  5 00:07:37 2015 kern.info /usr/bin/ss-tunnel[1533]: [udp] cache hit: 8.8.4.4:53 <-> 127.0.0.1:32871
```

oglops, 2015-11-05:
进程还在用perp这样的监视进程工具有意义么？ 我手动装了，但是觉得好像没意义吧，perp又不知道是不是该重启进程（如果进程还在）
https://ilmvfx.wordpress.com/2015/10/12/using-perp-to-supervise-shadowsocks-daemon-on-tomato-router/

closed 2015-11-21 by aa65535

qifuren1985, 2016-11-29:
如果我在路由器上面开启ss-redir, 然后在一台pc上开启ss-local, 两个连的同一个ss, 这样就会导致路由器上面的ss-redir无法连上, pc上面的ss-local倒是正常. 如果关掉pc上面的ss-local, 路由器上面的ss-redir就又正常了.