shadowsocks/openwrt-shadowsocks #107: 通过openwrt开发分支，编译错误

hero18688, 2015-07-22:
提示如下，请解决
"mips-openwrt-linux-uclibc-gcc" GCC="mips-openwrt-linux-uclibc-gcc" CXX="mips-openwrt-linux-uclibc-g++" RANLIB="mips-openwrt-linux-uclibc-gcc-ranlib" STRIP=mips-openwrt-linux-uclibc-strip OBJCOPY=mips-openwrt-linux-uclibc-objcopy OBJDUMP=mips-openwrt-linux-uclibc-objdump SIZE=mips-openwrt-linux-uclibc-size CROSS="mips-openwrt-linux-uclibc-" ARCH="mips" ;
make[4]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2'
make  all-recursive
make[5]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2'
Making all in libsodium
make[6]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium'
Making all in dist-build
make[7]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/dist-build'
make[7]: Nothing to be done for `all'.
make[7]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/dist-build'
Making all in msvc-scripts
make[7]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/msvc-scripts'
make[7]: Nothing to be done for `all'.
make[7]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/msvc-scripts'
Making all in src
make[7]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/src'
Making all in libsodium
make[8]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/src/libsodium'
Making all in include
make[9]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/src/libsodium/include'
make[9]: Nothing to be done for `all'.
make[9]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/src/libsodium/include'
make[9]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/src/libsodium'
make[9]: Nothing to be done for `all-am'.
make[9]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/src/libsodium'
make[8]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/src/libsodium'
make[8]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/src'
make[8]: Nothing to be done for `all-am'.
make[8]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/src'
make[7]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium/src'
make[7]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium'
make[7]: Nothing to be done for `all-am'.
make[7]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium'
make[6]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libsodium'
Making all in libcork
make[6]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libcork'
make[6]: Nothing to be done for `all'.
make[6]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libcork'
Making all in libipset
make[6]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libipset'
make[6]: Nothing to be done for `all'.
make[6]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libipset'
Making all in libudns
make[6]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libudns'
make[6]: Nothing to be done for `all'.
make[6]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libudns'
Making all in libev
make[6]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libev'
make[6]: Nothing to be done for `all'.
make[6]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/libev'
Making all in src
make[6]: Entering directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/src'
  CCLD     ss-local
/home/workingdir/Openwrt/981213_150712/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib/libcrypto.so: warning: gethostbyname is obsolescent, use getnameinfo() instead.
../libsodium/src/libsodium/.libs/libsodium.a(libsodium_la-stream_chacha20_ref.o): In function `chacha_encrypt_bytes.part.0':
stream_chacha20_ref.c:(.text+0x1a0): undefined reference to `__stack_chk_guard'
stream_chacha20_ref.c:(.text+0x9b8): undefined reference to `__stack_chk_guard'
stream_chacha20_ref.c:(.text+0x9d0): undefined reference to `__stack_chk_fail'
stream_chacha20_ref.c:(.text+0xa18): undefined reference to `__stack_chk_guard'
../libsodium/src/libsodium/.libs/libsodium.a(libsodium_la-stream_chacha20_ref.o): In function `crypto_stream_chacha20_ref_xor_ic':
stream_chacha20_ref.c:(.text+0xbb4): undefined reference to `__stack_chk_guard'
stream_chacha20_ref.c:(.text+0xd64): undefined reference to `__stack_chk_fail'
../libsodium/src/libsodium/.libs/libsodium.a(libsodium_la-xor_salsa20_ref.o): In function `crypto_stream_salsa20_xor_ic':
xor_salsa20_ref.c:(.text+0xc): undefined reference to `__stack_chk_guard'
xor_salsa20_ref.c:(.text+0x25c): undefined reference to `__stack_chk_guard'
xor_salsa20_ref.c:(.text+0x274): undefined reference to `__stack_chk_fail'
../libsodium/src/libsodium/.libs/libsodium.a(libsodium_la-auth_poly1305_donna.o): In function `crypto_onetimeauth_poly1305_donna':
auth_poly1305_donna.c:(.text+0x9b4): undefined reference to `__stack_chk_guard'
auth_poly1305_donna.c:(.text+0xa50): undefined reference to `__stack_chk_fail'
../libsodium/src/libsodium/.libs/libsodium.a(libsodium_la-verify_poly1305_donna.o): In function `crypto_onetimeauth_poly1305_donna_verify':
verify_poly1305_donna.c:(.text+0x14): undefined reference to `__stack_chk_guard'
verify_poly1305_donna.c:(.text+0x70): undefined reference to `__stack_chk_fail'
collect2: error: ld returned 1 exit status
make[6]: *** [ss-local] Error 1
make[6]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2/src'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/workingdir/Openwrt/981213_150712/build_dir/target-mips_34kc_uClibc-0.9.33.2/shadowsocks-libev/openssl/shadowsocks-libev-2.2.2'
make[4]: *** [all] Error 2
[... cut ...]
make[3]: Leaving directory `/home/workingdir/Openwrt/981213_150712/package/my_package/shadowsocks-libev'
make[2]: *** [package/my_package/shadowsocks-libev/compile] Error 2
make[2]: Leaving directory `/home/workingdir/Openwrt/981213_150712'
make[1]: *** [/home/workingdir/Openwrt/981213_150712/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/workingdir/Openwrt/981213_150712'
make: *** [world] 错误 2
make j=5 V=99  44.33s user 13.28s system 53% cpu 1:48.38 total
➜  981213_150712 git:(master) ✗

closed 2015-07-22 by aa65535

hero18688, 2015-07-22:
怎么关闭了 没有回复

wongsyrone, 2015-07-22:
重新编译工具链，用官方默认musl

hero18688, 2015-08-06:
不可以用uclibc编译了吗