shadowsocks/openwrt-shadowsocks #109: v2.2.3 无法工作

braveguywallce, 2015-07-23:
shadowsocks_spec_polarssl_ar71xx.ipk, v2.2.3, 在opnwrt BARRIER BREAKER (14.07, r42625) 下安装，首先是下列提示：
Installing shadowsocks-libev-spec-polarssl (2.2.2-1) to root...
Configuring shadowsocks-libev-spec-polarssl.
Configuring xsocks.
//usr/lib/opkg/info/xsocks.postinst: line 4: default_postinst: not found
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/shadowsocks is different from the conffile in the new package. The new conffile will be placed at /etc/config/shadowsocks-opkg.
 * pkg_run_script: package "xsocks" postinst script returned status 127.
 * opkg_configure: xsocks.postinst returned 127.

xsocks配置出错。

然后在Luci里跑不起来，进入命令格式：
/etc/init.d/shadowsocks start
提示
/etc/rc.common: line 1: server: parameter not set or null

重新安装回v.2.2.2-1后，一切正常工作，请问这是什么原因呢？谢谢！

aa65535, 2015-07-24:
应该是系统的 base-files 版本太旧，shadowsocks 与 xsocks 也没有关系。
至于 「postinst returned 127.」错误，你可以按照 aa65535/openwrt-dnsmasq#3 中的方法临时解决一下。

closed 2015-07-24 by aa65535

Crystommy, 2015-07-24:
实测无法运行是/etc/config/shadowsocks配置文件更改所致，需要使用新版luci-app-shadowsocks

commit 31ad31cf (2015-07-24): fixed #109