shadowsocks/openwrt-shadowsocks #118: 2.2.2-1版本没有生成访问控制对应的iptables

catscarlet, 2015-07-29:
版本：shadowsocks-libev-spec_2.2.2-1_ramips_24kec.ipk

/etc/config/shadowsocks

```
config shadowsocks
        option tunnel_enable '1'
        option tunnel_port '5300'
        option tunnel_forward '8.8.4.4:53'
        option use_conf_file '1'
        option config_file '/etc/shadowsocks/config.json'
        option lan_ac_mode '1'
        list lan_ac_ip '192.168.1.105'
        option enable '1'
        option ignore_list '/etc/shadowsocks/ignore.list'
        option udp_relay '1'
```

运行/etc/init.d/shadowsocks start之后，生成了对应的ipset

```
Name: ss_spec_lan_ac
Type: hash:net
Revision: 4
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 8536
References: 0
Members:
192.168.1.105 nomatch
```

但是没有生成对应的iptalbes条目。导致luci界面上的访问控制功能失效，所有流量都变成走ss判断逻辑了（虽然有忽略列表但是如果底下设备比较复杂的话不太方便）。

之前版本装的shadowsocks-libev-spec_2.2.1-1_ramips_24kec.ipk，走的貌似是iptables直接添加逻辑？不是ipset逻辑？

因为没有2.2.3-1的预编译版本，所以没有测试最新版。

catscarlet, 2015-07-29:
尝试了2.2.1版本，shadowsocks直接启动不能

```
root@OpenWrt:/usr/bin# /etc/init.d/shadowsocks start
ipset v6.20.1: Unknown argument: `nomatch'
Try `ipset help' for more information.
```

刚才diff了2.2.1和2.2.2的代码，没有修改这里。看release log，应该2.2.3也没有修改。等下我去diff一下2.1.4的，应该是改了tproxy逻辑时把这里忽略了吧。

不过标准的openwrt版里面，是没有“访问控制”功能的，只有装了这个项目里给的那个luci界面之后才有，其他用户应该没有感知。

aa65535, 2015-07-30:
duplicate with #77

新版可以在这里下载 http://openwrt-dist.sourceforge.net/releases/

closed 2015-07-30 by aa65535

catscarlet, 2015-07-30:
与 #77 没什么关系。#77 只是2.2.1运行rules时ipset hash:ip不支持nomatch而已。我昨晚通宵看过近期的所有issue了。
刚才试了shadowsocks-libev-spec_2.2.3-1_ramips_24kec.ipk和luci-app-shadowsocks-spec_1.3.3-1_all.ipk

目前现象是：

- 设置为仅允许列表内，则所有流量都走ss，无视列表设置

```
Name: ss_spec_lan_ac
Type: hash:net
Revision: 4
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 8536
References: 1
Members:
192.168.1.199 nomatch
```

- 设置为仅允许列表外，则所有流量都不走ss，无视列表设置

```
Name: ss_spec_lan_ac
Type: hash:net
Revision: 4
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 8536
References: 1
Members:
192.168.1.199
```

- 不论设置允许列表外还是允许列表内，iptables都没什么变化

```
# Generated by iptables-save v1.4.21 on Thu Jul 30 16:00:05 2015
*nat
:PREROUTING ACCEPT [546:31479]
:INPUT ACCEPT [567:32179]
:OUTPUT ACCEPT [95:6431]
:POSTROUTING ACCEPT [9:866]
:SS_SPEC_WAN_AC - [0:0]
:SS_SPEC_WAN_FW - [0:0]
:delegate_postrouting - [0:0]
:delegate_prerouting - [0:0]
:postrouting_lan_rule - [0:0]
:postrouting_rule - [0:0]
:postrouting_wan_rule - [0:0]
:prerouting_lan_rule - [0:0]
:prerouting_rule - [0:0]
:prerouting_wan_rule - [0:0]
:zone_lan_postrouting - [0:0]
:zone_lan_prerouting - [0:0]
:zone_wan_postrouting - [0:0]
:zone_wan_prerouting - [0:0]
-A PREROUTING -j delegate_prerouting
-A OUTPUT -p tcp -m comment --comment _SS_SPEC_RULE_ -j SS_SPEC_WAN_AC
-A POSTROUTING -j delegate_postrouting
-A SS_SPEC_WAN_AC -m set --match-set ss_spec_wan_ac dst -j RETURN
-A SS_SPEC_WAN_AC -j SS_SPEC_WAN_FW
-A SS_SPEC_WAN_FW -p tcp -j REDIRECT --to-ports 56345
-A delegate_postrouting -m comment --comment "user chain for postrouting" -j postrouting_rule
-A delegate_postrouting -o br-lan -j zone_lan_postrouting
-A delegate_postrouting -o pppoe-wan -j zone_wan_postrouting
-A delegate_prerouting -m comment --comment "user chain for prerouting" -j prerouting_rule
-A delegate_prerouting -i br-lan -j zone_lan_prerouting
-A delegate_prerouting -i pppoe-wan -j zone_wan_prerouting
-A zone_lan_postrouting -m comment --comment "user chain for postrouting" -j postrouting_lan_rule
-A zone_lan_postrouting -s 192.168.1.0/24 -d 192.168.1.102/32 -p tcp -m tcp --dport 8091 -m comment --comment "pi-8091 (reflection)" -j SNAT --to-source 192.168.1.1
-A zone_lan_prerouting -m comment --comment "user chain for prerouting" -j prerouting_lan_rule
-A zone_lan_prerouting -s 192.168.1.0/24 -d 222.128.138.1/32 -p tcp -m tcp --dport 8091 -m comment --comment "pi-8091 (reflection)" -j DNAT --to-destination 192.168.1.102:8091
[... 86 lines cut ...]
-A zone_lan_forward -j zone_lan_dest_ACCEPT
-A zone_lan_input -m comment --comment "user chain for input" -j input_lan_rule
-A zone_lan_input -m conntrack --ctstate DNAT -m comment --comment "Accept port redirections" -j ACCEPT
-A zone_lan_input -j zone_lan_src_ACCEPT
-A zone_lan_output -m comment --comment "user chain for output" -j output_lan_rule
-A zone_lan_output -j zone_lan_dest_ACCEPT
-A zone_lan_src_ACCEPT -i br-lan -j ACCEPT
-A zone_wan_dest_ACCEPT -o pppoe-wan -j ACCEPT
-A zone_wan_dest_REJECT -o pppoe-wan -j reject
-A zone_wan_forward -m comment --comment "user chain for forwarding" -j forwarding_wan_rule
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "Accept port forwards" -j ACCEPT
-A zone_wan_forward -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "user chain for input" -j input_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment Allow-DHCP-Renew -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment Allow-Ping -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "Accept port redirections" -j ACCEPT
-A zone_wan_input -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "user chain for output" -j output_wan_rule
-A zone_wan_output -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i pppoe-wan -j reject
COMMIT
# Completed on Thu Jul 30 16:00:05 2015
```

diff过一次2.1.4到2.2的代码，这里相关的改动太大了，没理解，也没尝试成功修复。

aa65535, 2015-07-30:
@catscarlet 我测试过访问控制，但无法重现你的问题。

```
-A zone_lan_prerouting -p tcp -m set ! --match-set ss_spec_lan_ac src -m comment --comment _SS_SPEC_RULE_ -j SS_SPEC_WAN_AC
```