shadowsocks/openwrt-shadowsocks #119: 使用OpenVPN连接到OpenWRT后，无法翻墙

wingerted, 2015-08-06:
如题，现在的情况是，直接连接路由器是可以走通翻墙的，但是使用openVPN连接是不行，怀疑是iptables的设置有问题？

clowwindy, 2015-08-06:
我记得 OpenVPN 的 NAT 是它自己做的，不是走的系统，这样它有可能走的 INPUT chain。

wingerted, 2015-08-06:
所以原来是走的POSTROUTING所以OK，现在这个项目的规则走的是INPUT，所以OpenVPN走不过来了对吧。。。

aa65535, 2015-08-07:
@wingerted 如果不需要走 INPUT 的话，可以修改 /etc/init.d/shadowsocks 的[这个](https://github.com/shadowsocks/openwrt-shadowsocks/blob/master/files/shadowsocks.spec#L60)地方，去掉 `-o`

```diff 
@@ -57,7 +57,7 @@ start_rules() {
                -b "$wan_bp_ip" \
                -w "$wan_fw_ip" \
                -e "$ipt_ext" \
-               -o $udp
+               $udp
        return $?
 }
```

closed 2015-08-07 by aa65535

mysteriouss, 2016-05-20:
@aa65535 如果openvpn proto是tcp的，请问如何设置iptables，查到以下 

```bash
# This file is interpreted as shell script.
# Put your custom iptables rules here, they will
# be executed with each firewall (re-)start.

# Allow all traffic in and out of the tun interface.
iptables -A input_rule      -i tun+ -j ACCEPT
iptables -A output_rule     -o tun+ -j ACCEPT
# This rule will allow traffic towards internet from tun
iptables -A forwarding_rule -i tun+ -j ACCEPT
iptables -A forwarding_rule -o tun+ -j ACCEPT
```
部分只被DNS污染的网站可以访问，谷歌和脸书还是不能

samtry, 2016-09-26:
@mysteriouss  请问怎么添加一级路由，碰到同样的问题。路由brctl addif br-lan tun0添加失败。

Biercenter, 2016-09-26:
我用最新版2.5.3，安装openvpn后一切正常了，能正常翻墙，你更新下新版的吧！我记得2.5.0以后就可以了。

samtry, 2016-09-27:
@Biercenter  路由器是openwrt mt7620，没有找到2.5以上的。

Biercenter, 2016-09-27:
我也是mt7620路由器，自己通过imagebuild编译的最新版lede，再用sdk编译一个最新版的openwrt-shadowsocks就可以了。说说我现在的方案吧，lede（openwrt）+shadowsocks+dnsmasq（gwflist通过shadowsocks查询）+openvpn=一切正常，如果用chinadns替代gwflist，openvpn就无法解析到ip地址，至于为什么会这样我没排查。

mysteriouss, 2016-09-29:
@samtry 再买一个路由器..