shadowsocks/openwrt-shadowsocks #122: 每次路由器重启后都需要手动restart

wangxj-nemo, 2015-08-10:
shadowsocks-libev-spec 2.2.3-1版本，每次路由器重启发现shadowsocks都没有自己启动，需要手动restart，

补充信息提供：
启动脚本里面chinadns和shadowsocks优先级都是90，shadowsocks处于启用状态
本地启动脚本里面加入/etc/init.d/shadowsocks restart也不行。

Dreista, 2015-08-22:
请问是否使用了 sqm？
上次编译的时候额外加入了 sqm-scripts 和 luci-app-sqm 之后出现了这种情况。
目前路由器没有在身边，不好做测试，没法确定是不是它们的锅_(:з」∠)_

wangxj-nemo, 2015-08-22:
不好意思，不太明白，你是问我有没有安装sqm吗？

Dreista, 2015-08-22:
是的。

wangxj-nemo, 2015-08-24:
没装，opkg list和进程里面都没有，另外我的DLINK的WLAN口也是需要插拔网线或者LUCI里面点击“连接”才能获得地址

Dreista, 2015-08-24:
那应该不是这个原因了。
我更改无线设置并保存后也需要重启。
AR71XX, WNDR3800, Openwrt Chaos Calmer RC3

wangxj-nemo, 2015-08-24:
现在原因找不到，那能不能在启动进程里面加入ss启动的功能，还有wlan点击连接触发的代码又是什么，请教如何加这2个功能

aa65535, 2015-08-24:
@wangxj-nemo 
接口重新连接后会执行 /etc/hotplug.d/iface/ 下面的脚本。
这个应该是固件的问题了。接口的高级设置里面有没有勾选开机自动运行？

closed 2015-08-24 by aa65535

wangxj-nemo, 2015-08-24:
应该不是固件的原因，因为装ss之前一切正常，不需要插拔网线；装过ss和chinadns后就这样了。