shadowsocks/openwrt-shadowsocks #123: IPV6支持

athanoslee, 2015-08-10:
我使用的是教育网是原生 DHCP IPV6。我的想法是VPS和OPENWRT之间在IPV6上使用SS，全局透明代理。然后其他所有设备通过OPENWRT免配置翻墙（我的nexus 6连到交换机的时候，并不能使用play store里的SS客户端连到VPS的IPV6，故想通过路由器利用IPV6）。然后按照@cokebar 的[这篇博客](https://cokebar.info/archives/664)设置了SS（我使用的是2.2.2 spec版，openwrt BB 14.07），此时已经可以通过路由器上网了，但是有V6上的DNS污染的现象。

无论是设置UDP转发，还是PDNSD，结果都是CHROME 报错DNS PROBED FINISHED NO INTERNET。但是我的配置应该是正确的，因为两种方式使用DHCP IPV4和VPS的V4地址的时候都是通的（虽然速度很着急）（PDNSD的时候还正确地产生了缓存文件）。

如果说UDP不通是因为UDP会受到主动的干扰（但是我在PC上用插网线和SSPC端的时候，没管DNS也可以访问所有的网站，包括youtube视频等UDP服务，这是什么原理？），那为什么TCP转发也不能成功呢？是不是SS没有自动把PDNSD发的包转走（V4时就是好的啊），还是中间V4-V6的时候出现了故障（那未使用hosts又为什么可连）？还是除了SS地址一栏其他哪里V4，V6下的设置有所不同我没有注意？

并不是计算机专业人士，若描述或所遇情况显得幼稚请海涵。如果有什么测试手段能更清楚地将情况呈现出来也请不吝赐教。

aa65535, 2015-08-19:
shadowsocks 和服务端的连接可以使用 ipv6 ，但 ipv6 地址将不会被 shadowsocks 代理。
所以，如果 ipv6 地址是被封锁的，将打不开。
至于 DNS 污染，可以使用 ss-tunnel 解决。

closed 2015-08-19 by aa65535

athanoslee, 2015-09-12:
最后解决了：
最后是pdnsd将dns请求以tcp的方式发给ss转发给google dns。
用的是ss的udp转发功能，但是实际上转发的是tcp。
应该是ss没有把dns查询自动转走（就是说通过dnsmasq或者pdnsd直接发到8888时，无论是tcp还是udp ss都并没有转走这个信息。），必须通过打开udp转发功能把dns喂给ss转发给8888，而且用udp不通，应该是被墙了（所以dnsmasq不行）。而且直接送到1080端口也不行，因为这样发送的dns查询并没有指定发送给谁，只能借助ss的udp转发功能，因为里面本身有写发给8888的位置。

IMconnor, 2017-06-18:
> 但 ipv6 地址将不会被 shadowsocks 代理。所以，如果 ipv6 地址是被封锁的，将打不开。

能不能改成代理ipv6地址？因为这两天以来出现了https://github.com/lennylxx/ipv6-hosts/issues/125 描述的问题，亚太的Youtube视频服务器会“403 Forbidden”来自大陆地区的ipv6访问，服务端搞事情奈我何……教育网主机在路由器正确配置了ipv6支持的情况下会优先通过ipv6直连Youtube，结果是看不了视频。如果想走代理只能开PC版的Shadowsocks或者禁用ipv6协议，但是这样就上不了教育网ipv6站啦（比如六维）。

cokebar, 2017-06-18:
@IMconnor v4/v6双栈时候会复杂很多  没有原生双栈环境来测试不好弄，如果路由是双栈，服务器只有ipv4就更蛋疼了。

cokebar, 2017-06-18:
@IMconnor 比较舒服的做法是，在chinadns/dns-forwarder之类的dns转发器那里弄个功能，抛掉国外网站的AAAA record

IMconnor, 2017-06-18:
@cokebar 兴许我弄错了，不管是在本地主机还是在VPS上，也不管是"-4"还是"-6"，都是下面的结果，除非绕开香港的路由，即router不含hkg https://redirector.googlevideo.com/report_mapping
否则没办法看Youtube视频了。所以r2---sn-npoeen76这样的节点到底出啥问题，多大仇…嗯关于你说的丢掉aaaa结果我在网络属性勾掉ipv6协议证明是好用的。

~$ curl -6 -v 'https://r2---sn-npoeen76.googlevideo.com/videoplayback?ip=47.52.*.*&lmt=1497652990941960&expire=1497738125&clen=439569668&source=youtube&id=o-ACoptxfD86q8O-q5XYSOY-s40_8rYInyxvd8UsKlllNX&dur=868.900&mime=video%2Fwebm&gir=yes&signature=546B2FE04E5FC299C711D8745F2943FC462C8762.63C1781423D9FA7FDEDE79C5422C601562EDCAE3&ei=LVdFWfK9CI7c4gKxtZHIDA&requiressl=yes&itag=303&sparams=clen,dur,ei,expire,gir,id,initcwndbps,ip,ipbits,itag,keepalive,lmt,mime,mm,mn,ms,mv,pl,requiressl,source&keepalive=yes&ipbits=0&pl=21&key=cms1&alr=yes&ratebypass=yes&cpn=d5cJ6pceBubit9e3&c=WEB&cver=1.20170615&cms_redirect=yes&mm=26&mn=sn-npoeen76&ms=onr&mt=1497716385&mv=u&range=0-1272143&rn=8&playerretry=1&rbuf=0'
.
.
.
HTTP/1.1 403 Forbidden
Last-Modified: Wed, 02 May 2007 10:26:10 GMT
Content-Type: text/plain
Content-Length: 0
Connection: close
X-Content-Type-Options: nosniff
Date: Sun, 18 Jun 2017 06:10:23 GMT
Server: gvs 1.0

Closing connection 0

cokebar, 2017-06-18:
@IMconnor 这两家U2B炸了，阿里云直接看不了，其他的也是需要关了QUIC或者开启UDP代理才行。
话说那个链接即使你pc能下载，放到server上也可能是403，因为那个都是带了身份认证的，有IP和MAC的hash之类的指纹信息，换台机子那个链接是无效的

IMconnor, 2017-06-18:
@cokebar 原来是这样，那还是不折腾静静地等Youtube自己修。