shadowsocks/openwrt-shadowsocks #127: 当前版本不支持最新内核，报错，请解决，感谢！

mapleshadow, 2015-08-19:
Installing shadowsocks-libev-spec (2.3.0-1) to root...
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipt-conntrack and kmod-ipt-conntrack) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipv6 and kmod-ipv6) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ip6tables and kmod-ip6tables) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (kmod-ipt-core and kmod-ipt-core) providing same name marked HOLD or PREFER. Using latest.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for shadowsocks-libev-spec:
 *      kernel (= 3.18.11-1-c9110281555cccc9dd13e092c2acaeaf) *         kernel (= 3.18.11-1-c9110281555cccc9dd13e092c2acaeaf) *         kernel (= 3.18.11-1-c9110281555cccc9dd13e092c2acaeaf) *
 * opkg_install_cmd: Cannot install package shadowsocks-libev-spec.

以上是错误信息，openwrt版本号是：
OpenWrt Chaos Calmer 15.05-rc3 / LuCI Master (git-15.167.71950-8005ecb)

mapleshadow, 2015-08-19:
话说回来，shadowsocks-libev_2.3.0-1_ar71xx.ipk可以正常安装，但是shadowsocks-libev_2.3.0-1_ar71xx.ipk这个版本没有luci的gui界面，不方便。有luci gui界面的版本shadowsocks-libev-spec又无法安装。。。。。。。。。

aa65535, 2015-08-19:
看一下opkg源是不是在 `https://downloads.openwrt.org/chaos_calmer/15.05-rc3/` 下.
安装之前使用 `opkg update` 更新一下

mapleshadow, 2015-08-20:
源是：
http://downloads.openwrt.org/chaos_calmer/15.05-rc1/ar71xx/ 下。
做了 opkg update 更新
之后再次安装，问题依旧，应该是还不支持15.05-rc1的新的内核导致的问题。

aa65535, 2015-08-20:
这个错误是依赖需要的 kmod 与内核版本不一致导致的。
实在不行的话重刷固件。

mapleshadow, 2015-08-20:
是的，我现在正在根据15.05-rc1的源代码重新编译该ipk，就提个建议把，官方也可以考虑跟进openwrt的版本号较好，一般openwrt规矩高版本直接兼容低版本。

aa65535, 2015-08-20:
跟ss软件包无关，是你的软件源里的 kmod 版本和你固件的内核不匹配。

https://github.com/shadowsocks/openwrt-shadowsocks/issues/127#issuecomment-132902346

mapleshadow, 2015-08-21:
按照你的意思，我应该如何做？

aa65535, 2015-08-21:
查看一下现在的固件的 kernel 版本是否为 `3.18.17-1-5dcb66d04a38219f7fd91c748f0e9a32`，不是的话，需要重新刷对应版本的固件。
如果固件是从官方源下载的，应该不会出现这个问题。
出现这个问题的一般是自己编译的固件。

closed 2015-08-21 by aa65535

mapleshadow, 2015-08-22:
刷固件无效，仍然是报错内核不匹配。
最终，我用openwrt源代码，导入shadowsocks源代码，重新编译，编译出来的ipk再安装，就OK了。
PS：另外路由环境内的源配置，也重新修改成与当前内核版本一致的网址路径。

mie1987, 2015-12-21:
我的也是这样的错误，装shadowsocks-libev-spec_2.4.3-1_ramips_24kec.ipk运行也是报Cannot open session to kernel错误。openwrt15.05，内核3.18.20，已经核对过kmod了，怎么解决啊，不想自己编译啊。

chor, 2015-12-29:
我也是跟楼上一样，装shadowsocks-libev-spec_2.4.3-1_ramips_24kec.ipk运行也是报Cannot open session to kernel错误。openwrt15.05，内核3.18.20。小米路由mini，今天才装的。

chaogu, 2017-04-03:
我重新执行一下opkg update可以了，我是netgear wndr4300

debiansid, 2017-04-03:
You should compile whole thing by yourself