shadowsocks/openwrt-shadowsocks #130: Openwrt 客户端运行一段时间问题

sinalpha, 2015-08-21:
shadowsocks 在openwrt上运行一段时间后，会有一个诡异的现象：

ssh 登陆路由器，在路由器（wan）上执行`wget`, `curl`, `ping`等命令，可以和代理服务器通信，能够正常访问获取墙外资源。
但是，在路由器的客户端（br-lan）workstation，无法和代理服务器通信。

必须执行 `\etc\init.d\shadowsocks restart`才能使客户端访问代理服务器。

---

尝试过 `\etc\init.d\chinadns restart` , `\etc\ini.t\network restart` , `\etc\init.d\dnsmasq restrat`均不能消除这个现象。

---

所以问题是，无法在openwrt路由器上监控shadowsocks客户端是否正常运行。

感觉shadowsocks在openwrt上运行一段时间后存在一个`伪正常`（面向路由器正常工作，面向主机不正常）的运行状态。

aa65535, 2015-08-21:
使用 iptables-save 打印一下出问题时的规则。

sinalpha, 2015-08-24:
@aa65535  谢谢，根据[第5016行的内容](https://github.com/sinalpha/diff/commit/79dbb8e4707233c07df8b858c06565f898721d36)

我手动添加
`-A zone_lan_prerouting -p tcp -m comment --comment _SS_SPEC_RULE_ -j SS_SPEC_LAN_AC`

然后， `iptables-restore` 问题就解决了。路由器不用重启shadowsocks，pc端就可以恢复翻墙了。

---

为什么openwrt运行一段时间后，防火墙的规则会自动少了一条呢？和shadowsocks有关系吗？

aa65535, 2015-08-24:
中间有没有做什么操作？这个是 zone_lan_prerouting 链被重置导致的。
将 zone_lan_prerouting  改为 PREROUTING 应该可以解决。

sinalpha, 2015-08-25:
没有做任何特别的操作，这个个规则自己就掉了。

昨天手动加上后，到今天为止，就没有再掉过。

不清楚为什么。

xiaolanlou, 2015-09-01:
你好，我是小米路由器没有openwrt官方的sdk，只有小米发布的交叉编译工具链，请问如何编译？

closed 2015-09-17 by aa65535

lincank, 2015-09-28:
怎么revert了？这个问题还是没有解决？

aa65535, 2015-09-29:
@lincank 会导致内存占用过多，另外丢失 zone_lan_prerouting 并不常见，可视为 OpenWrt 自身BUG。

jack338c, 2015-10-26:
我手动添加
-A zone_lan_prerouting -p tcp -m comment --comment _SS_SPEC_RULE_ -j SS_SPEC_LAN_AC
@sinalpha ,我也有这样的问题,请问这一步具体怎么改?

commit 3087a54b (2015-09-17): fix #130
commit 30f8ceac (2015-09-20): Revert "fix #130"

This reverts commit 3087a54bda6a3a99d0c4974a850c016546629b6b.