shadowsocks/openwrt-shadowsocks #141: Chaos Calmer 15.05 不能從軟件源下載

fadequ, 2015-09-18:
你好，15.05 不能使用這個軟件源

	src/gz openwrt_dist http://openwrt-dist.sourceforge.net/releases/ar71xx/packages
	src/gz openwrt_dist_luci http://openwrt-dist.sourceforge.net/releases/luci/packages

使用 opkg update 命令收到如下報錯

	Collected errors:
	 * opkg_download: Failed to download http://openwrt-dist.sourceforge.net/releases/ar71xx/packages/Packages.sig, wget returned 8.
	 * opkg_download: Failed to download http://openwrt-dist.sourceforge.net/releases/luci/packages/Packages.sig, wget returned 8.

致謝！

aa65535, 2015-09-20:
`option check_signature 1`  change to `option check_signature 0` in `/etc/opkg.conf`

closed 2015-09-20 by aa65535

NemoAlex, 2015-09-21:
I have changed the `option check_signature 1` to `option check_signature 0`, but still has same error.
This is my /etc/opkg.conf:
```
dest root /
dest ram /tmp
lists_dir ext /var/opkg-lists
option overlay_root /overlay
option check_signature 0
src/gz chaos_calmer_base http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/nand/packages/base
src/gz chaos_calmer_luci http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/nand/packages/luci
src/gz chaos_calmer_packages http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/nand/packages/packages
src/gz chaos_calmer_routing http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/nand/packages/routing
src/gz chaos_calmer_telephony http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/nand/packages/telephony
src/gz chaos_calmer_management http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/nand/packages/management
src/gz openwrt_dist http://openwrt-dist.sourceforge.net/releases/ar71xx/packages
src/gz openwrt_dist_luci http://openwrt-dist.sourceforge.net/releases/luci/packages
```

Any suggestion?

NemoAlex, 2015-09-21:
Got it.
Removed the line `option check_signature 1` or `option check_signature 0`, it works fine.

aa65535, 2015-09-21:
you are right.

johnxo, 2016-05-20:
我也不能使用，签名改成0也不行
Downloading http://openwrt-dist.sourceforge.net/releases/ar71xx/packages/Packags.gz.
wget: can't connect to remote host (216.34.181.96): Connection timed out