shadowsocks/openwrt-shadowsocks #143: Chaos Calmer 15.05 WNDR4300 安装 polarssl 版本后无法启动

NemoAlex, 2015-09-21:
在 luci 里开启，刷新以后还是没有运行。找了一下进程也没有起来。
没有找到任何 log。
libpolarssl 版本 1.3.11-1。
卸载，安装 openssl 版本正常。

cielpy, 2015-09-22:
为什么要装polarssl版本。。4300这个内存够用吧。。装openssl版本不是更好吗。

NemoAlex, 2015-09-22:
是 http://openwrt-dist.sourceforge.net/ 自动安装脚本装上了 polarssl 的版本

cielpy, 2015-09-22:
你可以下载ipk安装openssl版本，一样的。
opkg install shadowsocks-libev-spec 这个应该装的是openssl版本吧，我在WNDR4300上一直用的挺正常的。

-- 
beijiu572-gmail
Sent with Airmail

NemoAlex, 2015-09-22:
我知道 openssl 版本可以用。但是这里需要讨论的是 polarssl 版本的问题。
@cielpy  谢谢你的帮助。

cielpy, 2015-09-22:
运行不起来看看端口有没有占用，有没有写日志，有的话看有没有对文件的写入权限，暂时只能想到这里。
我觉得吧，既然openssl版本能用，为什么要折腾一个除了加密方式其他功能没有任何区别的版本呢，况且polarssl版本的加密方式还是比较少的那一个。
-- 
beijiu572-gmail
Sent with Airmail

aa65535, 2015-09-22:
手动执行 `/etc/init.d/shadowsocks start` 查看报错信息。
我这里重现的是 `can't load library 'libpolarssl.so.7'`
说明符号链接不正确
可以执行下面两条命令修复一下
```sh
cd /usr/lib/
ln -s libmbedtls.so libpolarssl.so.7
```

closed 2015-09-22 by aa65535