shadowsocks/openwrt-shadowsocks #145: shadowsocks-libev-spec & chinadns 在15.05版本上安装后卸载报错。

pupie, 2015-09-23:
如下所示，ss-libev-spec和chinadns ipk安装包和15.05版本固件同时编译，ss和chinadns未集成到固件中。安装后再卸载会报错

root@Gate:/tmp# opkg remove shadowsocks-libev-spec
Removing package shadowsocks-libev-spec from root...
You can force removal of packages with failed prerm scripts with the option:
        --force-remove
No packages removed.
Collected errors:
 * pkg_run_script: package "shadowsocks-libev-spec" prerm script returned status 1.
 * opkg_remove_pkg: not removing package "shadowsocks-libev-spec", prerm script failed

aa65535, 2015-09-23:
无法重现。

closed 2015-09-23 by aa65535

JayvicWen, 2015-10-10:
@aa65535 表示我也这样了……

我是 WNDR4300，然后从 http://openwrt-dist.sourceforge.net/releases/ar71xx/packages 下载的 ipk，然后 `opkg install shadowsocks-libev-spec_2.4.0-1_ar71xx.ipk` 装的，卸载就会和 @pupie 一样……

```
root@OpenWrt:~/software# opkg remove shadowsocks-libev-spec
Removing package shadowsocks-libev-spec from root...
You can force removal of packages with failed prerm scripts with the option:
	--force-remove
No packages removed.
Collected errors:
 * pkg_run_script: package "shadowsocks-libev-spec" prerm script returned status 1.
 * opkg_remove_pkg: not removing package "shadowsocks-libev-spec", prerm script failed
```

ZaXk, 2016-01-07:
遇到同样问题！
由于第一次安装出现了问题，好像需要安装某个依赖软件时少了某个文件的，当时没记录下来。想卸载重装再找找原因。但卸载就出现问题，强制卸载之后，安装，没有出现第一次的错误，再卸载还是出错。

root@OpenWrt:~# opkg install shadowsocks-libev-spec
Installing shadowsocks-libev-spec (2.4.3-1) to root...
Downloading http://openwrt-dist.sourceforge.net/releases/mvebu/packages/shadowsocks-libev-spec_2.4.3-1_mvebu.ipk.
Configuring shadowsocks-libev-spec.
root@OpenWrt:~# opkg remove shadowsocks-libev-spec
Removing package shadowsocks-libev-spec from root...
You can force removal of packages with failed prerm scripts with the option:
        --force-remove
No packages removed.
Collected errors:
 * pkg_run_script: package "shadowsocks-libev-spec" prerm script returned status 1.
 * opkg_remove_pkg: not removing package "shadowsocks-libev-spec", prerm script failed