shadowsocks/openwrt-shadowsocks #147: 最新的2.3.3-1版无法通过配置文件的方式启动

maoyuanrun, 2015-09-29:
路由器安装的shadowsocks-libev-spec_2.3.3-1_ramips_24kec，通过luci app在网页上填写好服务器ip、端口等信息，保存&应用，shadowsocks能够正常启动运行。但是，如果勾选“使用配置文件"，则显示“shadowsocks未运行”，尝试通过命令 /etc/init.d/shandowsocks start 启动，却报“/etc/rc.common: line 1: server: parameter not set or null”的错误。。。确定配置文件/etc/shadowsocks/config.json无误，是从/var/etc/shadowsocks.json复制过去的，删除了local_address这一项，上一版也是这么偷懒干的，并且能够正常启动。

aa65535, 2015-09-29:
更新 [luci-app-shadowsocks-spec](https://sourceforge.net/projects/openwrt-dist/files/luci-app/shadowsocks-spec/)

closed 2015-09-29 by aa65535

maoyuanrun, 2015-09-29:
luci-app是最新的luci-app-shadowsocks-spec_1.3.2-1_all，截图http://imgjar.com/g/a1vtu0em54

aa65535, 2015-09-29:
新版已经不支持“使用配置文件"

blog2i2j, 2015-09-29:
啥时候出2.4.0版啊?好像更新了很多啊

aa65535, 2015-09-29:
明天

https://github.com/shadowsocks/openwrt-shadowsocks/issues/147#issuecomment-144103715

blog2i2j, 2015-09-29:
感谢感谢. aa还更新dnsmasq吗? 好像很长时间不更新了耶

aa65535, 2015-09-29:
直接到 openwrt 的官方源下载安装最新版就可以，patch已经合并进去了。

https://github.com/shadowsocks/openwrt-shadowsocks/issues/147#issuecomment-144105349

blog2i2j, 2015-09-29:
另外问一下, 最新的2.3.3的ss是不是不支持填写服务器域名了? 只能写IP地址?

blog2i2j, 2015-09-29:
啥时候aa能够搞个dnsmasq的tcp查询patch呢? 现在搭个dnsmasq还得另外搭个tcp 的dns, 一个能搞定该多好啊?

flipphos, 2015-09-29:
现在运行只能得到ss server（国外）ip，导致外网无法连接plex server（大陆ip）。将plex
server运行的电脑加入luci中的忽略列表也无效。路由器上端口已经映射。
困扰了我好久的问题。请指点一下，谢谢！

ohyeah521, 2015-09-30:
@aa65535  @blog2i2j  我也再考虑如何让dnsmasq支持 tcp dns查询，装了pdnsd ，非常占用内存。如果有好的方法请分享一下。

xseven007, 2015-10-07:
@ohyeah521 redsocks支持返回让前置程序利用tcp协议重新进行DNS解析的方法。