shadowsocks/openwrt-shadowsocks #151: 安装2.4.0-1，luci-app-shadowsocks-spec，shadowsocks-libev-spec-polarssl报错，显示是安装完了，但不能运行。

hubgitt, 2015-10-02:
安装2.4.0-1，luci-app-shadowsocks-spec，shadowsocks-libev-spec-polarssl报错，显示是安装完了，但不能运行。
Package luci-app-shadowsocks-spec (1.3.6-1) installed in root is up to date.
Configuring shadowsocks-libev-spec-polarssl.
//usr/lib/opkg/info/shadowsocks-libev-spec-polarssl.postinst: line 4: default_postinst: not found
Collected errors:
 * pkg_run_script: package "shadowsocks-libev-spec-polarssl" postinst script returned status 127.
 * opkg_configure: shadowsocks-libev-spec-polarssl.postinst returned 127.
------------------------------------------------------
Package shadowsocks-libev-spec-polarssl (2.4.0-1) installed in root is up to date.
Configuring shadowsocks-libev-spec-polarssl.
//usr/lib/opkg/info/shadowsocks-libev-spec-polarssl.postinst: line 4: default_postinst: not found
Collected errors:
 * pkg_run_script: package "shadowsocks-libev-spec-polarssl" postinst script returned status 127.
 * opkg_configure: shadowsocks-libev-spec-polarssl.postinst returned 127.

aa65535, 2015-10-03:
固件太旧了

closed 2015-10-03 by aa65535

c1024, 2015-10-16:
我也是出现这个问题，我用的是Openwrt 14.07官方版。这个固件也太旧？

aa65535, 2015-11-04:
编辑 `/lib/functions.sh` 这个文件在末尾加上
```sh
default_postinst() {
	return 0
}
```

flyher, 2015-11-11:
@aa65535
同样报这个问题。
但是软件包管理中显示已经更新到2.4，
同时更新了luci，
重启路由，然而并未见到一次性验证。