shadowsocks/openwrt-shadowsocks #153: Is it ok about "Connection reseted by peer"?

lixingcong, 2015-10-05:
There are so many reseted by peers in my openwrt system log.But there is no problem visiting overseas websites.
What do these logs mean? 
Would my host be fucked by GFW?

    Wed Sep 30 11:25:18 2015 authpriv.info dropbear[1494]: Exit (root): Disconnect received
    Wed Sep 30 11:28:58 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:28:58 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:28:58 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:29:46 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:29:46 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:29:46 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:29:46 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:29:46 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:29:46 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:29:46 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:29:46 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer
    Wed Sep 30 11:30:00 2015 cron.info crond[840]: USER root pid 1570 cmd echo 3 > /proc/sys/vm/drop_caches
    Wed Sep 30 11:30:00 2015 kern.info kernel: [  935.210000] sh (1570): drop_caches: 3
    Wed Sep 30 11:30:57 2015 kern.err /usr/bin/ss-redir[1402]: server recv: Connection reset by peer

aa65535, 2015-10-10:
no problem

closed 2015-10-10 by aa65535

rogeryan, 2015-11-30:
I am wondering why this log was shown too?

tedaz, 2016-02-17:
What is the meaning of this log?