shadowsocks/openwrt-shadowsocks #158: ss-tunnel和UDP Relay咨询

JayvicWen, 2015-10-11:
没太弄明白 dns 解析应该怎么配，所以来咨询下……

打开 UDP Relay 的话，路由器下面连接的设备 dns 是自动就好了么，我打开了，然后电脑 nslookup、dig 都还是不行：

```
➜  (Jayvic-MBP) ~  dig baidu.com @192.168.1.1                                                                                                                                           [10/11/15 16:04:44]

; <<>> DiG 9.8.3-P1 <<>> baidu.com @192.168.1.1
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: REFUSED, id: 41393
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;baidu.com.			IN	A

;; Query time: 2 msec
;; SERVER: 192.168.1.1#53(192.168.1.1)
;; WHEN: Sun Oct 11 16:04:51 2015
;; MSG SIZE  rcvd: 27
```

然后 ss-tunnel 就是把指定端口的请求转到设定的 ip:port 上么，我也开了，然后 5300 端口转到 10.131.1.19:5300（转到的这个地方是内网里自己用 pdnsd 开的 dns 解析，因为只有那台机器能上网，路由本身是没有外网的），本机 dig 也不行：

```
➜  (Jayvic-MBP) ~  dig baidu.com @192.168.1.1 -p 5300                                                                                                                                   [10/11/15 16:06:00]

; <<>> DiG 9.8.3-P1 <<>> baidu.com @192.168.1.1 -p 5300
;; global options: +cmd
;; connection timed out; no servers could be reached
```

直接 dig 10.131.1.19:5300 是可以正常解析的。

然后最后我在 `/etc/dnsmasq.conf` 中设置 `server=10.131.1.19#5300`，本机就可以正常解析上网了（但是 ping 还是不通的，目测是 udp 都不通）。如果设成  `server=127.0.0.1#5300` 也不行：

```
➜  (Jayvic-MBP) ~  dig baidu.com @192.168.1.1                                                                                                                                           [10/11/15 16:12:50]

; <<>> DiG 9.8.3-P1 <<>> baidu.com @192.168.1.1
;; global options: +cmd
;; connection timed out; no servers could be reached
```

我刷的 WNDR4300 原配 openwrt，里面木有 dig 的样子，所以没在路由上面 dig 过……

现在已经完全懵了，想请教下这两个东西到底是干嘛的，以及该怎么用才对……

还有顺便问下，由于路由本身没外网，然后我其实想弄的是，一般是直接走 ss服务器1（速度快，未翻墙） 上外网，然后墙外列表中国年的走 ss服务器2（速度慢点，但翻墙了），这个如何改 ss-rules 是好……

aa65535, 2015-10-11:
UDP Relay 不支持 lo 转发，路由本身的UDP不会被代理，所以需要在客户端自定义 DNS 地址。
推荐做法是使用 ss-tunnel 提供 DNS 的查询，转发地址的 ip:port 是相对于 ss 服务端的，所以填写 10.x.x.x 的地址会无效。

closed 2015-10-11 by aa65535

JayvicWen, 2015-10-11:
饿，还是没太明白，路由本身的 UDP 不会被代理不是说在路由上进行 DNS 查询不行么，为啥是需要在客户端自定义 DNS。
然后 ss-tunnel 的话，好像现在理解了点，之前似乎完全理解错了，但是我 ss 服务端也是同一个 10.x.x.x，所以相对于 ss 服务端应该也可以吧，然后就算填写 114 或者 Google 的 DNS 也不行，但在 ss 服务端所在那台服务器上，直接都是可以解析的…… 至今为止，我没能通过 ss-tunnel 解析 DNS 成功过，咋设置都跪……