shadowsocks/openwrt-shadowsocks #163: shadowsocks 开机不启动问题

wangxj-nemo, 2015-10-22:
1， 启动脚本里面shadowsocks都是enable和start的。

2，每次重新上电都需要通过dropbear进路由器手动键入/etc/init.d/shadowsocks start才能启动ss-redir

3, 即便在rc.local文件里面加入
sleep 20
/etc/init.d/shadowsocks start也不行。

后面看到shadowsocks和chinadns处于同一启动优先级90，于是改/etc/rc.d下面的S90shadowsocks文件名为S92shadowsocks和链接到的/etc/init.d/shadowsocks文件的START=92，并注释STOP=15,还是不能解决开机不启动的问题。

附加信息，/etc/rc.d下面有2个链接K15shadowsocks和S90shadowsocks指向同一个文件/etc/init.d/shadowsocks，开机后进程里面有{S90shadowsocks} /bin/sh /etc/rc.common /etc/rc.d/S90shadowsocks boot的进程，就是没有ss-redir的进程，需要手动输入。

请教如何解决，谢谢！！

wangxj-nemo, 2015-10-22:
经过折腾发现：
K15shadowsocks和S90shadowsocks这2个符号链接文件跟openwrt luci里面的启动项shadowsocks的启用/禁用是关联的，删除上面2个文件显示为已禁用；但是进程里面出现想要的ss-redir；界面里面启用shadowsocks，会自动生成上面2个符号链接；

得出的结论：luci界面里面的启动脚本的启用/禁用，和2个链接文件的生成，shadowsocks开机启动无效，也和rc.local文件冲突。

closed 2015-11-21 by aa65535

liuyindyx, 2015-11-29:
我的也是这个问题，我是小白，请问这个问题怎么解决。每次重启路由器都得手动start。