shadowsocks/openwrt-shadowsocks #164: OpenWrt SDK编译ss时候出现如下问题

dlinkqu, 2015-10-27:
Package shadowsocks-libev-spec is missing dependencies for the following libraries:
libcrypto.so.1.0.0
libpthread.so.0
libz.so.1

搜索发现staging_dir\target-mips_34kc_uClibc-0.9.33.2\usr\lib下面有这几个库文件，不知道哪里有问题。

dlinkqu, 2015-10-27:
已解决。

closed 2015-10-27 by dlinkqu

ivyswen, 2016-01-24:
请问如何解决的？

dlinkqu, 2016-02-02:
staging_dir\target-mips_34kc_uClibc-0.9.33.2\pkginfo 下面，没有libcrypto libpthread libz，需要到openwrt对应目录找到包含缺失依赖的文件复制到SDK目录下：

cp openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/libopenssl.provides OpenWrt-SDK-ar71xx-for-redhat-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2_o/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo                     --------for libcrypto&libz
cp  openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/libpthread.provides OpenWrt-SDK-ar71xx-for-redhat-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2_o/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo                      --------for libpthread