shadowsocks/openwrt-shadowsocks #172: ss-rules report error

smilekevin, 2015-11-16:
even with latest script in master after fix #171 , still getting error below

root@Buffalo:/usr/bin# /bin/sh /usr/bin/ss-rules -s <someip> -l 1080 -S  -L  -i /etc/ignore.list -a b192.168.1.162 -b  -w  -e  -o
Bad argument `set'
Try `iptables -h' or 'iptables --help' for more information.

environment:
Openwrt: chaos_calmer/15.05

aa65535, 2015-11-16:
Can not reproduce, ipset installed?

smilekevin, 2015-11-16:
Yes, 

root@Buffalo:~# ipset -V
ipset v6.24, protocol version: 6
root@Buffalo:~# iptables -V
iptables v1.4.21

root@Buffalo:~# /bin/sh /usr/bin/ss-rules -s 45.79.110.34 -l 1080 -S -L -i /etc/ignore.list -a b192.168.1.162 -b -w -e -o
Bad argument `set'
Try `iptables -h' or 'iptables --help' for more information.
root@Buffalo:~# 

I added echo in script and the error comes from following iptables command:

root@Buffalo:~#  iptables -t nat -A zone_lan_prerouting -p tcp -o -m set ! --match-set ss_spec_lan_ac src -m comment --comment _SS_SPEC_RULE_ -j SS_SPEC_WAN_AC
Bad argument `set'
Try `iptables -h' or 'iptables --help' for more information.

aa65535, 2015-11-17:
`-S` and `-L`  requires an argument. eg: `-S 45.79.110.34 -L 1080`

smilekevin, 2015-11-17:
See my previous comment I put IP and port in the command. The very first comment I was trying to hide the IP.

smilekevin, 2015-11-17:
also if I ran following command direct it fails as well. 

root@Mercury:~# iptables -t nat -A zone_lan_prerouting -p tcp -o -m set ! --matc
h-set ss_spec_lan_ac src -m comment --comment SS_SPEC_RULE -j SS_SPEC_WAN_AC
Bad argument 'set'
Try 'iptables -h' or 'iptables --help' for more information.

aa65535, 2015-11-17:
Should be like this: 
`/bin/sh /usr/bin/ss-rules -s 45.79.110.34 -l 1080 -i /etc/ignore.list -a b192.168.1.162 -o`

closed 2015-11-17 by aa65535

smilekevin, 2015-11-17:
If ipt_ext is missing in the config. then it will generate command like /usr/bin/ss-rules ...  -e -o

	/usr/bin/ss-rules \
		-s "$server" \
		-l "$local_port" \
		-S "$udp_server" \
		-L "$udp_local_port" \
		-i "$ignore_list" \
		-a "$ac_args" \
		-b "$wan_bp_ip" \
		-w "$wan_fw_ip" \
		-e "$ipt_ext" \
		-o $udp

smilekevin, 2015-11-17:
and so does $udp

aa65535, 2015-11-17:
see `ss-rules -h`
empty string is still a parameter value.

smilekevin, 2015-11-18:
This command was not generated by me but the system. I just managed to capture the command issued when i click "Save and apply" on luci. 

I guess this is because of following code in shadowsocks.spec

	/usr/bin/ss-rules \
		-s "$server" \
		-l "$local_port" \
		-S "$udp_server" \
		-L "$udp_local_port" \
		-i "$ignore_list" \
		-a "$ac_args" \
		-b "$wan_bp_ip" \
		-w "$wan_fw_ip" \
		-e "$ipt_ext" \
		-o $udp

It should detect if $ip_ext or $udp is set. If not the command does become "-e -o"

smilekevin, 2015-11-18:
since this one is closed. I am opening "#173". I still think this is an issue.