shadowsocks/openwrt-shadowsocks #173: detect if $ipt_ext or $udp is set in shadowsocks.spec

smilekevin, 2015-11-18:
in shadowsocks.spec
/usr/bin/ss-rules \
    -s "$server" \
    -l "$local_port" \
    -S "$udp_server" \
    -L "$udp_local_port" \
    -i "$ignore_list" \
    -a "$ac_args" \
    -b "$wan_bp_ip" \
    -w "$wan_fw_ip" \
    -e "$ipt_ext" \
    -o $udp

It should detect if $ip_ext or $udp is set. If not the command does become "-e -o"

aa65535, 2015-11-18:
No, it will become `-e "" -o`, this will not be wrong.

closed 2015-11-18 by aa65535

halftan, 2016-03-26:
add with these lines in /etc/init.d/shadowsocks

```
        echo \
        /usr/bin/ss-rules \
                -s "$server" \
                -l "$local_port" \
                -S "$udp_server" \
                -L "$udp_local_port" \
                -a "$ac_ips" \
                -i "$(uci_get_by_type access_control wan_bp_list)" \
                -b "$(uci_get_by_type access_control wan_bp_ips)" \
                -w "$(uci_get_by_type access_control wan_fw_ips)" \
                -e "$(uci_get_by_type access_control ipt_ext)" \
                -o $ARG_UDP
```

and the result is 

```
/usr/bin/ss-rules -s 10.0.0.1 -l 1080 -S  -L  -a  -i /etc/myroute_bypasslist.txt -b  -w  -e  -o -u
```
which is the same line echoed with `set -x`