shadowsocks/openwrt-shadowsocks #174: 只安装luci-app-shadowsocks-spec和shadowsocks-spec就上不了网了

cuteribs, 2015-11-20:
新装的openwrt, 安装了shadowsocks-spec和luci-app-shadowsocks-spec, 启动后客户端就上不了网了. 重启openwrt也不行. 关闭后又能上网了.

google了好久也没见有什么详细点的使用说明 -_ -, 能找到的全部都是和chinadns一起用的..

光装这2个不够使用透明代理么?

aa65535, 2015-11-21:
默认启动是全局代理，如果不能上网，请检查是否正确配置。

closed 2015-11-21 by aa65535

cuteribs, 2015-11-21:
我不用luci-app-shadowsocks-spec,只用shadowsocks-spec手动运行ss-redir, 再手动跑iptables配置.是可以实现透明代理的.

newboydj169, 2015-11-30:
有同样的问题，只要一启动shadowsocks-spec就不行。
但额外启动chinadns就可以上网。但这时感觉国内网站访问变慢.

liuyindyx, 2015-11-30:
我想问一下shadowsocks的代理原则是哪里控制的？ignor.list应该怎么配置，放在什么位置。我现在所有的都走shadowsocks代理。并且每次重启路由器后还得手动命令启动shadowsocks和chinadns，在luci配置界面中，startup都是开启的。
chinadns应该只是负责返回正确的DNS接卸结果而已，真正和服务器通不通过代理还是应该有shadowsocks来负责吧。

newboydj169, 2015-12-14:
代理控制原理取决于你的方案。
+ 是GFWLIST+SS的方案，那就是
1. 通过dnsmasq，将需要翻墙的host送到你自己的DNS解析防止污染。 解释出来的IP，放入到gfwlist这个ipset中。例：比如你在dnsmasq的配置目录的某个文件中写入：
`server=/.lsxszzg.com/127.0.0.1#5300`
`ipset=/.lsxszzg.com/gfwlist`
在此例中，是我自定义的DNS转发服务器127.0.0.1:5300.
2. 在用户自定的防火墙规则中,指定将所有gfwlist中的IP，全部转发给SS的代理端口，比如我的是1080，地是写成：
`ipset -N gfwlist iphash`
`iptables -t nat -A PREROUTING -p tcp -m set --match-set gfwlist dst -j REDIRECT --to-port 1080`
- 第一行，建立gfwlist这个ipset.
- 第二行，将gfwlsi中的IP全部转发给SS的代理端口。

+ 其他方案请google.

newboydj169, 2015-12-14:
刚搜了下资料，同时我猜了下，ss-spec版是启动的ss-redir，然后用了ss-rules这个脚本生成了一张代理规则表，也即是更智能，使用户只需装上ss不作任何配置即可翻墙。但某些情况，我说不清楚的情况，他将一些国内网站加入了这张规则表中，但由于ss-redir启动后由于各种原因不能代理，所以上不了网，比如远程服务器不可用等。
所以，我在/etc/init.d/shadowsocks中，将ss-rule的启动关闭了。然后还是用回以前手动的方案。