shadowsocks/openwrt-shadowsocks #177: 是否支持配置多服务器自动切换

Vigi0303, 2015-11-27:
总用一个服务器配置有时候会有连不上的情况。
能不能像windows版客户端一样弄个自动切换的功能。

seanlvjie, 2015-11-30:
严重支持

hdstar2009, 2015-12-03:
不用自动切换，有的时候ping值好不一定就是想要的线路。
只要有个能设置多个账号，选择后手工切换就方便了。免得每次换账号要重新输入。
luci不会写。

aa65535, 2015-12-03:
自动切换目前无法去做, 多配置切换倒是可以, 不过目前比较忙, 等有空吧.

hdstar2009, 2015-12-04:
期待。有多个ss账号的就方便多了。

javenfeng, 2015-12-20:
我是把不同服务器的配置文件写到/ect/shadowsocks文件夹中，起个简短的名字，然后勾选使用配置文件，需要哪个服务器就手动把.conf前的名称改掉就行，也比较方便。

hdstar2009, 2015-12-21:
我用的是spec版，没那个使用配置文件的选择了。

另外最近有看到人宣传什么ss plus，还分什么lite standard
pro三个版本，看了宣传觉得没什么变化（虽然写了一堆资料看起来很高大上），除了目前没见到的（现在提供的都是lite版本，根据gfwlist和自定义域名和ip来fq）pro据说有多ss账号切换功能。

希望楼主能增加多账号手工切换这个方便功能。

closed 2015-12-23 by aa65535

hdstar2009, 2015-12-24:
会这两天出pre-release还是自己可以覆盖你修改过的文件来测试？

aa65535, 2015-12-24:
@hdstar2009 等 `shadowsocks-libev` 更新吧, 也可以覆盖有改动的文件.
不过要注意一点: `/etc/config/shadowsocks` 文件需要改成跟新版结构一样,否则 luci 显示不正常.

chenxinxing, 2015-12-30:
多服务器配置文件shadowsocks.config，这个要如何添加多个服务器信息呢？ 具体如何使用呢？

aa65535, 2015-12-31:
@chenxinxing 需要同时更新 luci-app

chenxinxing, 2016-01-14:
@aa65535 q1：是否只有spec版本支持多服务器的么？ q2: 如果不安装luci-app，配置文件如何添加多个服务器信息呢？

beautifulape, 2016-02-15:
如果自动切换不方便做，是否能增加一个“ss启动时随机使用多个服务器配置中的一个”这样的可选项？因为之前看到@cokebar 写了一个定时检测ss连接状态并根据情况重启的脚本。不知道这样的话实现起来是否更容易一些。

非常感谢！

commit 2c8d01ab (2015-12-23): multi-server support & close #177

must use luci-app-shadowsocks-spec v1.3.8 or later