shadowsocks/openwrt-shadowsocks #185: 关于源代码中shadowsocks.include文件的疑问

kulike, 2015-12-20:
在我编译OPENWRT 14.07 过程中，发现编后的固件 路由器重启后 SS 系统进程有显示，但是就是不能出国，设置全局模式，百度IP 也显示本地地址，必须手动启动SS 才能一切正常，安装 那几个依赖包 也不行，后来结合 你这里的OPENVPN 源码，我把shadowsocks.include文件 复制并修改名称到了/etc/hotplug.d/iface/30-shadowsocks，然后SS重启不正常的现象没有了。
疑问？？
源文件定义的/usr/share/shadowsocks/firewall.include，在系统中为何失去了作用，用14.07官方源，除了加入LUCI和SS ，chindns 等几个文件改了固件大小外没有做过任何修改，是还有软件没有安装吗？

我把 shadowsocks.include 文件修改名称 移动到/etc/hotplug.d/iface/30-shadowsocks ，对系统有没有影响？

aa65535, 2015-12-20:
可以看一下 `/etc/config/firewall` 是否有 `shadowsocks` 节.

closed 2015-12-20 by aa65535

kulike, 2015-12-21:
忘了说明 路由器是 TP-LINK TL-WR941NV2，改 16M存储，64M内存 加USB
根据你的提示SS启动后 在/etc/config/firewall 中没有 shadowsocks 字节。
刚刚做了测试，官方版 14.07 固件，安装SS 启动后在/etc/config/firewall  也没有出现 shadowsocks 字节。路由器重启后SS工作不正常,每次必须手动启动才能正常使用。
15.05和trunk 官方固件，安装SS启动后，在/etc/config/firewall  出现 shadowsocks 字节。路由器重启后SS工作正常。
安装包均为 shadowsocks-libev-spec_2.4.3-1_ar71xx.ipk和luci-app-shadowsocks-spec_1.3.2-1_all.ipk
难道是兼容性问题？
又发现一个问题，SS 不正常的情况下 ，谷歌和脸书都不能访问，但是推特网页可以正常访问，开始以为是CHINDNS 的导致，关闭SS 重启路由器后，推特 网页 就打不开了。似乎SS 也在工作？

aa65535, 2015-12-21:
目前提供的 ipk 是 15.05 的SDK编译的, 旧版的 base-files 没有相应的函数去执行 `postinst-pkg`.
可以修改一下 `/lib/functions.sh` 添加下面的两个函数.
```sh
default_postinst() {
	local ret=0
	local root="${IPKG_INSTROOT}"
	local pkgname="$(basename ${1%.*})"
	if [ -f "$root/usr/lib/opkg/info/${pkgname}.postinst-pkg" ]; then
		( . "$root/usr/lib/opkg/info/${pkgname}.postinst-pkg" )
		ret=$?
	fi
	return $ret
}

default_prerm() {
	local name=$(basename ${1%.*})
	[ -f /usr/lib/opkg/info/${name}.prerm-pkg ] && . /usr/lib/opkg/info/${name}.prerm-pkg
	return 0
}
```