shadowsocks/openwrt-shadowsocks #187: tproxy + ss-redir 的Linux路由器不能正常工作。请分享经验

shifengjiang, 2015-12-21:
我一年多来使用自己搭建的 shadowsocks-libev + ChinaDNS 的Debian 7.x 路由器，没有遇到什么问题，期间shadowsocks-libev 升级到2.0.1。 
最近决定升级到另一台Debian 8.2的机器，并使用最新版的 shadowsocks-libev 2.4.3，为了先看看是否可用，仅仅使用了经过验证的config.json配置文件和官方的如下推荐命令：

  sudo iptables -t nat -A SHADOWSOCKS -d IP_of_my_remote_server -j RETURN

  sudo iptables -t nat -A SHADOWSOCKS -d 0.0.0.0/8 -j RETURN
  sudo iptables -t nat -A SHADOWSOCKS -d 10.0.0.0/8 -j RETURN
  sudo iptables -t nat -A SHADOWSOCKS -d 127.0.0.0/8 -j RETURN
  sudo iptables -t nat -A SHADOWSOCKS -d 169.254.0.0/16 -j RETURN
  sudo iptables -t nat -A SHADOWSOCKS -d 172.16.0.0/12 -j RETURN
  sudo iptables -t nat -A SHADOWSOCKS -d 192.168.0.0/16 -j RETURN
  sudo iptables -t nat -A SHADOWSOCKS -d 224.0.0.0/4 -j RETURN
  sudo iptables -t nat -A SHADOWSOCKS -d 240.0.0.0/4 -j RETURN

  sudo iptables -t nat -A SHADOWSOCKS -p tcp -j REDIRECT --to-ports 1080

  sudo ip rule add fwmark 0x01/0x01 table 100
  sudo ip route add local 0.0.0.0/0 dev lo table 100
  sudo iptables -t mangle -A SHADOWSOCKS -p udp --dport 53 -j TPROXY --on-port 1080 --tproxy-mark 0x01/0x01

  sudo iptables -t nat -A PREROUTING -p tcp -j SHADOWSOCKS
  sudo iptables -t mangle -A PREROUTING -j SHADOWSOCKS

  sudo ss-redir -u -c /etc/shadowsocks-libev/config.json -f /var/run/shadowsocks.pid

目前结果是不能访问任何外部网页。而在安装之前，我的这个Linux路由器是可以联网的 (通过PPPoE + iptables NAT)。

我注意到新版的ss-dir缺省监听地址是127.0.0.1而不再是曾经的0.0.0.0。请经验丰富的朋友帮助会诊。能传授一些诊断方法更是感激不尽！

heshiming, 2015-12-21:
我印象中是从 2.2.0 版开始 ss-redir 默认用 127.0.0.1 作为监听端口。要允许 iptables 转发，要加参数，如 `ss-redir -b 192.168.1.1` ，其余参数不变，其中 192.168.1.1 是路由 IP。

shifengjiang, 2015-12-21:
@heshiming 非常感谢你的提示，果然有效，无论是-b 192.168.1.1还是-b 0.0.0.0都可以。
第二个问题，即UDP的转发失效却依然存在，请问这个tproxy在普通linux发行版上有什么特别需要注意的地方吗？

closed 2015-12-25 by aa65535