shadowsocks/openwrt-shadowsocks #190: UDP forwarding detectable by black hats?

jchia, 2015-12-23:
Hi,

I've been using this for several months together with ChinaDNS. In the past few days, DNS query to the forwarded UDP port sometimes worked but often didn't (I got no response). I'm sure the peer on the other side of the wall is still working as I can do a DNS query from it to 8.8.4.4. In fact I have two geographically-separate servers and I get the same problem regardless of which one I connect to.

Have the wall builders found a way to detect (not necessarily decrypt) tunneled UDP traffic? I didn't change my config and I can't find any other reasonable explanation for the recent increase in DNS query failure rate.

How does UDP forwarding work in shadowsocks? It's tunneled in the TCP connection to the server, right?

rootless4real, 2015-12-23:
I can't say for certain in shadowsocks streams, but the Chinese are now
able to do DPI on encrypted openvpn UDP streams AND real-time packet
insertion in those streams.

heshiming, 2015-12-23:
ss-tunnel UDP relay is definitely experiencing trouble recently. It's very likely that these DNS queries are detectable due to their small size.

jchia, 2015-12-23:
If @heshiming is right, then maybe it would make sense for shadowsocks to have an option to obfuscate the UDP relay packets by tagging along some random-content random-length data when no tunneled TCP communication is happening. Except, I'm not sure where the defacto repo for shadowsocks is these days now that the @clowwindy one has been 'harmonized'.

mengskysama, 2015-12-23:
I face with the similar situation in the past few days ago.If the problem happen the same port TCP transfer is well but UDP pack loss.

heshiming, 2015-12-23:
As far as repo is concerned, the upstream of this openwrt version is still actively maintained: https://github.com/shadowsocks/shadowsocks-libev .

Although randomize UDP packet size might not help much, for them it's still possible to target a range of small sized packets. Unless of course, one decide to include up to 100 times as much garbage in the DNS query. Or add other options to compress that packet to under 10 bytes.

madeye, 2015-12-23:
I don't think the UDP packet loss is because of "Wall". Actually, it's a typical QoS issue observed this year for many ISPs in China.

To solve this, you'd try 
1. Buy some premium network service like CN2. As per my local tests, no UDP packet loss can be observed with CN2.
2. Try different ports. For example, set your UDP relay on the port 53 of your server.

heshiming, 2015-12-23:
Thank you for the pointers @madeye . When you say try different ports, do you mean to start another instance of shadowsocks on the server, on port 53, that is dedicated to UDP relay?

I'm on 国际精品网 although I do understand it's much worse than commercial CN2. This UDP trouble comes and goes. Sometimes it stuck for couple hours, and other times it remains good for the entire week. Once it's stuck, redialing the broadband connection helps, but only for minutes.

So it could be QoS.

madeye, 2015-12-23:
Right, openwrt-shadowsocks added this feature months ago: https://github.com/shadowsocks/openwrt-shadowsocks/issues/96

jchia, 2015-12-23:
I have a server running at port X. I just used iptables on the server to route ports 53 and 443 to port X:
```
sudo iptables -t nat -A PREROUTING -p tcp --dport 443 -j REDIRECT --to X
sudo iptables -t nat -A PREROUTING -p tcp --dport 53 -j REDIRECT --to X
```

So, it's effectively one server listening on ports X, 53 and 443. From an external perspective, this should be the same as running 3 servers on 3 ports.

On my openwrt router, I have the regular shadowsocks service running ```ss-redir``` and ```ss-tunnel```. I suppose the ss-tunnel is for the UDP forwarding. I manually start another ss-tunnel but connecting to port 53 on the server and receiving on another UDP port locally. I still get DNS query disruption.

So, "try different ports" didn't work, or maybe I'm missing something.

rootless4real, 2015-12-23:
I can confirm that trouble is most commonly experienced with DNS request
interception. I've gone to SSL tunneling (of openvpn) and am working on
integrating Shadowsocks with my router to try that out too.

The SSL tunneling is particularly effective.

aa65535, 2015-12-25:
@jchia 
Maybe you should use two `ss-server` to listening different port.
The UDP relay server use `-U` argument.

jchia, 2015-12-27:
Apparently, shadowsocks is not using TCP at all for the UDP-forwarding. I thought shadowsocks tunnels the DNS UDP packets through the TCP connection, but apparently it is sending the DNS UDP through UDP.

So, I also had to do the ```iptables``` ```PREROUTING``` with ```-p udp``` in order to use one ```ss-server``` process to serve on different ports for TCP and UDP. It's working now and I'm observing the reliability of the setup.

heshiming, 2015-12-29:
I'd like to report back that I've tried separating shadowsocks instances, with one dedicated instance for UDP relay, and the other for TCP only. The UDP relay port is lower than 53. In the last couple days, the UDP timeout problem appears to happen much less (before, it happened on a daily basis, couple times a day, now it happens probably one time every other day). It didn't go away completely though.

cokebar, 2015-12-30:
UDP connection is much less stable. As for DNS service, you may try tcp method, using eg. pdnsd

closed 2016-01-22 by aa65535