shadowsocks/openwrt-shadowsocks #196: b82f7c7以后的版本都无法运行

gyteng, 2016-01-02:
如题，`b82f7c7`之后的版本有修改过`/etc/init.d/shadowsocks`这个文件，在我的路由器上运行`/etc/init.d/shadowsocks boot`或者`/etc/init.d/shadowsocks start`均会报错，但我不知道哪里可以查看错误日志之类的，没法判断是什么原因造成的。

aa65535, 2016-01-02:
b82f7c7 只是 bug 修复, 既然 `/etc/init.d/shadowsocks start` 有报错, 可以将报错信息 post 上来.

gyteng, 2016-01-02:
是这样子的，我编译了b82f7c7版本可以正常使用，但是如果换成最新的ea10514版本，会出现这样的问题：

启动之后发现没有运行

    root@OpenWrt:/# ps | grep ss
    root@OpenWrt:/#

若手工运行的话，出现这种提示：

    root@OpenWrt:/# /etc/init.d/shadowsocks start
    Usage: ss-rules [options]

    Valid options are:

    -s <server_ip>          ip address of shadowsocks remote server
    -l <local_port>         port number of shadowsocks local server
    -S <server_ip>          ip address of shadowsocks remote UDP server
    -L <local_port>         port number of shadowsocks local UDP server
    -i <ip_list_file>       a file content is bypassed ip list
    -a <lan_ips>            lan ip of access control, need a prefix to
                            define access control mode
    -b <wan_ips>            wan ip of will be bypassed
    -w <wan_ips>            wan ip of will be forwarded
    -e <extra_options>      extra options for iptables
    -o                      apply the rules to the OUTPUT chain
    -u                      enable udprelay mode, TPROXY is required
    -U                      enable udprelay mode, using different IP
                            and ports for TCP and UDP
    -f                      flush the rules
    -h                      show this help message and exit
    root@OpenWrt:/#

aa65535, 2016-01-02:
因为在 2c8d01ab71df376e1d60c6ed14adc2c1070b3b19 的时候添加了多服务器的支持, 所以 `/etc/config/shadowsocks` 文件结构有所改变, 不能使用旧版的了, 可以尝试将旧版 `/etc/config/shadowsocks`  移除, 因为卸载或者直接更新是不会移除这个文件的.

closed 2016-01-02 by aa65535